kliment/Printrun #690: Windows 10 precompiled: AttributeError: 'Button' object has no attribute 'properties'

BillyBlaze, 2015-08-19:
With the latest precompiled build of 03-Feb-2015 I constantly get the following error when I click a custom button:

````  
File "printrun\pronterface.pyc", line 1939, in editbutton
AttributeError: 'Button' object has no attribute 'properties'
````

If I click a custom button and hold the button on his down state the error will flood the console until the state goes back to unpressed.

I also expect because of the above error that I cannot drag the custom buttons.

a4jp-com, 2016-01-04:
If you can compile the program can you make the files available. I really want to update from the old version on the website.

a4jp-com, 2016-01-04:
Or can you explain how to do the compiling? I can try the same thing as you if you like.

rockstorm101, 2017-12-15:
Hi @BillyBlaze, does this keep happening with the latest release?

BillyBlaze, 2017-12-15:
Hi @rockstorm101,

It still happens in the latest release. 
The line that is involved did change through:
```
Traceback (most recent call last):
  File "printrun\pronterface.pyc", line 2041, in editbutton
AttributeError: 'Button' object has no attribute 'properties'
```

My steps:
1. Download windows edition from releases
2. Extract files to Desktop
3. Open pronterface.exe
4. Create two buttons (Test2 and Test1)
5. Trying to drag Test1 result in the error above and doesn't change position.

rockstorm101, 2017-12-19:
Thanks. Reproduced.

 * OS: Windows 10 Pro
 * Printurn Version: 1.6.0

kliment, 2018-03-23:
This is now fixed in 2.x branch and will be in the next binary release

commit 9dea6417 (2018-03-23): Fix button dragging - fixes #690
closed 2018-03-23 by kliment