kliment/Printrun #691: request: jogging using arrow keys, pgup/pgdn

bubbapizza, 2015-08-26:
Just a suggestion to add a button that puts pronterface into a mode where you can jog using keyboard keys.

bubbapizza, 2015-08-26:
Or maybe using first-person shooter interface w/ the mouse and ASDW.  Firebutton starts extruding and scroll wheel jogs up and down on Z axis

kliment, 2018-03-27:
This already exists - click the little circles in the diagonal of the control buttons!

closed 2018-03-27 by kliment

VanessaE, 2018-05-09:
This does not work anymore - you can click the circle and get a tiny little D-pad icon, but when you press one of the up or down cursor keys, the input focus jumps from one widget to another (depending on which key you pressed) until it lands on a text field such as the X/Y feed rate or extrude speed, where it sticks (at which point, up/down increments or decrements the value, and left/right move the text cursor in the field).  Pressing right cursor while the focus is on the little D-pad does nothing at first, but holding it down for a second zooms in on the viewport.  Left cursor sometimes moves the X axis, or zooms out, or jumps between widgets, or just does nothing.