kliment/Printrun #694: pronterface dmg/binaries are out of date

queglay, 2015-10-04:
Currently, the pronterface binaries are out of date for mac.  This means that inexperienced new users on the mac will have trouble with the power saving features halting their prints.  perhaps it wont matter in the future so much, but the current mac .dmg files have serious issues.

it's not easy for newbies to go through the process of executing the git pronterface clone on mac, so I'd just like to request an update for that to help them out.

alf3, 2015-10-13:
+1 please

rockstorm101, 2017-12-15:
Hi @queglay, would you consider this matter solved with the new release?

queglay, 2017-12-15:
I'm unable to test because my printer is in another country unfortunately.
Thankyou for looking at this, I hope someone can test and comment...