kliment/Printrun #696: Cannot run from git repository on mac 10.10.5

queglay, 2015-10-13:
I've spent a few days trying to run the latest pronterface on my system.  the dependency issues are strange.  I cannot get the module Corefoundation in this version of osx.

```
Andrews-iMac:Printrun Audio$ ./pronterface.py 
WARNING:root:Memory-efficient GCoder implementation unavailable: dlopen(/Volumes/Data/Users/Audio/Dropbox/Open Civilisation/Github/Printrun/printrun/gcoder_line.so, 2): no suitable image found.  Did find:
	/Volumes/Data/Users/Audio/Dropbox/Open Civilisation/Github/Printrun/printrun/gcoder_line.so: mach-o, but wrong architecture
Traceback (most recent call last):
  File "./pronterface.py", line 32, in <module>
    from printrun.pronterface import PronterApp
  File "/Volumes/Data/Users/Audio/Dropbox/Open Civilisation/Github/Printrun/printrun/pronterface.py", line 32, in <module>
    from . import pronsole
  File "/Volumes/Data/Users/Audio/Dropbox/Open Civilisation/Github/Printrun/printrun/pronsole.py", line 42, in <module>
    from .power import powerset_print_start, powerset_print_stop
  File "/Volumes/Data/Users/Audio/Dropbox/Open Civilisation/Github/Printrun/printrun/power/__init__.py", line 21, in <module>
    from .osx import inhibit_sleep_osx, deinhibit_sleep_osx
  File "/Volumes/Data/Users/Audio/Dropbox/Open Civilisation/Github/Printrun/printrun/power/osx.py", line 20, in <module>
    import CoreFoundation
ImportError: No module named CoreFoundation
```

This is supposedly in the PyObjC library, but this cannot be installed in osx 10.10.5 for me either

```
Andrews-iMac:pyobjc-core-3.0.4 Audio$ python setup.py install
running install
running bdist_egg
running egg_info
writing include/pyobjc-compat.h to Lib/pyobjc_core.egg-info/include/pyobjc-compat.h
writing include/pyobjc-api.h to Lib/pyobjc_core.egg-info/include/pyobjc-api.h
writing Lib/pyobjc_core.egg-info/PKG-INFO
writing namespace_packages to Lib/pyobjc_core.egg-info/namespace_packages.txt
writing top-level names to Lib/pyobjc_core.egg-info/top_level.txt
writing dependency_links to Lib/pyobjc_core.egg-info/dependency_links.txt
reading manifest file 'Lib/pyobjc_core.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
warning: no directories found matching 'Scripts'
warning: no directories found matching 'setup-lib'
warning: no directories found matching 'source-deps'
warning: no previously-included files matching '*.pyc' found anywhere in distribution
warning: no previously-included files matching '*.pyo' found anywhere in distribution
warning: no previously-included files matching '*.so' found anywhere in distribution
warning: no previously-included files matching '*.dSYM/*' found anywhere in distribution
warning: no previously-included files matching '*.dSYM' found anywhere in distribution
warning: no previously-included files matching '__pycache__' found anywhere in distribution
writing manifest file 'Lib/pyobjc_core.egg-info/SOURCES.txt'
installing library code to build/bdist.macosx-10.5-fat/egg
running install_lib
running build_py
Overriding build_packages to copy PyObjCTest
running build_ext
error: Cannot locate a working compiler
```

rockstorm101, 2017-12-15:
Hi @queglay, did you manage to fix your problem? Are you able to run the latest Printrun release?

queglay, 2017-12-15:
Apologies, I'm unable to test because my printer is in another country

kliment, 2018-03-25:
I'm going to close this issue for now due to a lack of activity. If the problem occurs again feel free to reopen it.

closed 2018-03-25 by kliment