kliment/Printrun #697: Any patched versions or updates for Windows users?

a4jp-com, 2015-11-02:
Are there any patched versions or updates for Windows users? The new files look good but it would be nice if someone could compile the files so we can use them if possible.

tgit23, 2015-12-09:
Second that!

a4jp-com, 2015-12-10:
Whoops. I forgot I had written this.

tgit23, 2015-12-11:
I took the time to setup and run the recent release 20150310 from py source code and it's help-about box still says 2014-08-01.  Perhaps the compiled version at http://koti.kapsi.fi/~kliment/printrun/ really is the most recent release and there was just a mis-hap on updating the help-about box version label but really I have no idea.  I just used the  pre-compiled and updated slic3r.

rockstorm101, 2017-12-15:
I'm closing this issue because a new version has been released.

closed 2017-12-15 by rockstorm101