kliment/Printrun #700: Help needed: passing manual arguments to pronsole class

smsunny, 2015-11-04:
Hello,

I am a newbie to Printrun and python programming. I was trying to build a script that will pass manual arguments to 'pronsole' class without using cmd.
I am going to explain a little bit what I want. When I run pronsole and type 'connect' the software connects to the printer. Now I am trying to build an application on top of pronsole i.e. that will run pronsole in background. Now I am trying to pass the 'connect' command from my program, without using the cmd. 
What I have found in "pronsole.py" is that it calls the "cmdloop()" function to get command from cmd and use "parse_cmdline()" to start processing that command. Now I am trying to call the latter function like this:

interp = pronsole()
interp.parse_cmdline("connect")

But it generates error like this:

usage: test.py [-h] [-v] [-c CONF] [-e EXECUTE] [filename]
test.py: error: unrecognized arguments: o n n e c t

Certainly I am missing something. Can anyone please help me with this? Which function should I feed my command into? Or any other way to do what I have just described?

Thank you in advance

kliment, 2015-11-05:
parse_cmdline is meant to parse command-line arguments, not commands.
What you're looking for is interp.onecmd("connect")

smsunny, 2015-11-05:
Thank you very much. This is exactly what I have been looking for! :D

closed 2015-11-05 by smsunny

hadiboe, 2017-03-03:
Hi, May I know if say, I wanna use pronsole in the background but from Java, say by clicking a Jbutton to connect, is that possible?

hadiboe, 2017-03-03:
meaning - > using processbuilder -> pronsole command
snippet program:
String line;
				String linea;
				String lineb;
                OutputStream stdin = null;
                InputStream stderr = null;
                InputStream stdout = null;
                String pathToApplication3 = "C:/3D-Automatic/Printrun/pronsole.exe ";

	        	try {
        	        Runtime runTime = Runtime.getRuntime();   
        	        Process process = runTime.exec(pathToApplication3);       
        	        stdin = process.getOutputStream ();
        	        stderr = process.getErrorStream ();
        	        stdout = process.getInputStream ();
        	        
        	        linea ="connect" + "\n";
        	        stdin.write(linea.getBytes() );
        	        stdin.flush();