kliment/Printrun #701: RasberryPi2 and Multi-threading ?

JFPayeur, 2015-11-08:
I'v been using Pronterface for a long time on Windows xp and a Core2 Quad-core to print on a 
prusa i3. I feel like I'm trowing money out of the windows for electricity bill for this. Not to talk I want to get rid of Micro$oft 

I tried On a Rasberry Pi-1 B+ . I was just to slow 
I tried On a Rasberry Pi-2 . Still slow, especialy on loading a file, but still alot better  
I tried On a Linux Intelstick  . Still better , but a bit to slow for me, especially on loading a file again 

Got ubuntu on my big core2 quad, My best setup for now. 
But back on square one for power consumption 

One thing I find out is that the loading file is taking:  
 100%CPU on a RPi1  : 1Core , 
 25% on a Pi2 : 4Cores
 25% on a Intelstick : 4Cores 

Witch bring me the folowing conclusion: It is not multitreading 

Is there any plan to develop this ? RasberryPi2 are inexpesntive and would be a nice platform !

kliment, 2015-11-08:
It's probably the visualization that is the bottleneck here. See if
turning that off helps. The file parsing is indeed happening on a single
thread but it's independent of the gui thread.

ntoff, 2015-12-28:
A) it's 2015, enough of this "Micro$oft" bs
B) if it's a raspberry pi install octoprint and be done with it

kliment, 2018-03-25:
Unfortunately while we do use a separate thread for processing the file the GUI has to run on a single thread, and we're limited by that in this case. Disabling visualization should dramatically improve performance on your platform. Alternatively, you can use ntoff's suggestion and run octoprint, in which case the visualization is still single-threaded, but it slows down your browser instead of your pi.

closed 2018-03-25 by kliment