kliment/Printrun #703: Pronterface not booting; No attribute: 'gvis'

zappyguy, 2015-11-12:
Hi all,
I have a problem with pronterface. I have tried a few things here and there but it will not start. It was running well, with the exception triggering a memory dump every time it ran a print, and I decided to try and get it to run with Slic3r instead of Skeinforge to improve efficiency and to see if it would stop the memory dump happening. While it was working I:
- editted the script in options, forgot to change it back, it still ran. 
- redownloaded all its requirements and ran the gcoder_line fix, stopped running.
- after hours of searching the web, deleted the while file and reloaded it from git

This is the Error it now shows after giving:

NAME@PCNAME:~/Printrun$ ./pronterface.py 
/usr/share/themes/Lubuntu-dark-panel/gtk-2.0/apps/thunar.rc:55: error: invalid string constant "thunar-statusbar", expected valid string constant
Traceback (most recent call last):
  File "./pronterface.py", line 35, in <module>
    app = PronterApp(False)
  File "/home/NAME/Printrun/printrun/pronterface.py", line 2190, in __init__
    self.mainwindow = PronterWindow(self)
  File "/home/NAME/Printrun/printrun/pronterface.py", line 218, in __init__
    self.reload_ui()
  File "/home/NAME/Printrun/printrun/pronterface.py", line 280, in reload_ui
    self.settings.controlsmode == "Mini")
  File "/home/NAME/Printrun/printrun/gui/__init__.py", line 256, in createGui
    viz_pane = VizPane(self, vizpanel)
  File "/home/NAME/Printrun/printrun/gui/viz.py", line 108, in __init__
    if not isinstance(root.gviz, NoViz):
AttributeError: 'PronterWindow' object has no attribute 'gviz'

This netbook is an Aspire One running Lubuntu 15.04. half the ram is hardwired, half the ram is not hardwired and is not extendable beyond 1gb, I have looked around online and it is very annoying the ram is not expandable.

I will appreciate all the advice I can get.

Best Regards,
Geoff

zappyguy, 2015-11-12:
On a side note it was also giving this too for a while but stopped:
WARNING:root:RPC server bound on non-default port

rockstorm101, 2017-12-15:
Hi, @zappyguy, what is the status of this issue? Where you able to run Printrun normally?

zappyguy, 2017-12-16:
I have not seen this bug resurface. It seems to be running okay, although the laptop in question  is no more.

rockstorm101, 2017-12-17:
Good to hear, I'll be closing this issue then.

closed 2017-12-17 by rockstorm101