kliment/Printrun #705: Help needed: accessing current position of axes

smsunny, 2015-11-30:
Hello,

I am a newbie in regard of using Printrun. I have been trying to find a way to know the current position of X, Y and Z axis at anytime. I have been working with pronsole. I could not find any function or variable in 'printrun/pronsole.py' that returns current position of these axes. Can you please help me? Thank you in advance

kliment, 2015-11-30:
Send an M114 command to the printer and it will report its position

smsunny, 2015-11-30:
Thank you for the answer. Sending "M114" in the command line returns the positions. But this is exactly not the thing that I have been looking for. It is printing the output when I send the command. What I need is that it will return the positions as a variable which I can use for other calculations later. More elaborately speaking, maybe it will return a array containing the positions of x, y, z & e axis and I can use those array element later. Is there anyway to do that? Thanks in advance

closed 2015-12-08 by smsunny