kliment/Printrun #708: Printer disconnects when using network serial connection

WZ9V, 2015-12-17:
I'm using a Smoothie 5XC on my delta and have it attached via the Ethernet port.  Everything initializes and starts correctly but a few minutes into the print I get the following error.

ERROR] Can't write to printer (disconnected ?):
Traceback (most recent call last):
  File "/opt/Printrun/printrun/printcore.py", line 609, in _send
    self.printer.write(str(command + "\n"))
  File "/usr/lib/python2.7/socket.py", line 324, in write
    self.flush()
  File "/usr/lib/python2.7/socket.py", line 303, in flush
    self._sock.sendall(view[write_offset:write_offset+buffer_size])
timeout: timed out

The print continues for a while until the buffer runs out and then stops dead.  Thought it might be WiFi related so I attached everything to the switch and it still happens.

I'm using Linux Mint 17.3 Cinnamon and Pronterface 2015.03.10

cilynx, 2017-09-24:
This is because Smoothieware doesn't respond to TCP while the queue is full.  Pronterface has a hard-coded timeout of 0.25 seconds and tries 4-times before giving up and disconnecting, so if you have a single line of gcode that takes longer than 1-second, the connection will time out.  I changed the timeout in printcore.py to 5-minutes and haven't had any timeout disconnects since.

Discussion on Smoothieware project:

https://github.com/Smoothieware/Smoothieware/issues/872

Timeout setting in printcore.py:

https://github.com/kliment/Printrun/blob/f6605d61dae4a84f569db1f8c16c35127232ad0d/printrun/printcore.py#L197

rockstorm101, 2017-12-14:
Althoug a workaround was given, this issue raises some concerns for me:

 * Should this timeout be raised for all users? What would be the implications of a too high value here?
 
 * Should it not be hardcoded and made dependant of some configuration option instead?
 
Thoughts? (CC @kliment, @iXce)

kliment, 2017-12-14:
The implication of a too high value is that the machine sits around doing nothing until the host has figured out that something has gone wrong. This can be a problem if it is, say, in the middle of a print, where the nozzle will melt the surrounding area. The workaround is necessary in this case, and I would not be opposed to making it configurable, but I don't think it's justified to make it 5min for all users.

rockstorm101, 2017-12-15:
Alright then, I'll leave it open and label it as a request for implementing an option.