kliment/Printrun #709: Pronteface Click X or Y axis and both move diagonal, is this supposed to happen if not how do I fix it???

MazinDajani, 2015-12-18:
Fried the mother board on my Replicator 2x with a bad end stop. swapped the motherboard for a arduino and ramps, got it mostly working, Trying to test every thing in Pronteface and the X and Y axis run together, if I click x home it runs diagonal, slams into the y endstop then keeps grinding until it hits the x. which leads me to believe this is not a feature.

rockstorm101, 2015-12-19:
Hi Mazin,

No, I don't think that is supposed to happen like that, when clicking "Home X" only the X axis should move. Or at least this is how it works for me on a RepRap.

In https://github.com/kliment/Printrun/issues/702 kliment says that printrun is "intentionally incompatible [with Makerbot]". This being said, this seems to be a firmware problem, and therefore it would not concern this forum. The "Home X" button should be sending the command "G28 X", which homes the X axis. Some firmware, such as Delta printer's, are known to home every axis regardless of the axis given to the "G28" command [1]. So:
- Could you please tell what firmware you have installed on your arduino+ramps?
- Could you please check whether the command "G28 X" is indeed the command being used when you home the X axis?

[1] http://reprap.org/wiki/G-code#G28:_Move_to_Origin_.28Home.29

MazinDajani, 2015-12-19:
#define COREXY 
This was the culprit, sorry figured it out last night.

MazinDajani, 2015-12-19:
Thanx guys for your help

closed 2015-12-19 by MazinDajani