kliment/Printrun #710: Windows error message when loading file

chuck767, 2015-12-29:
Getting an error when loading  gcode in Pronterface (Printrun 2014.08.01)

This is the error message:

"windows error: exception: access violation reading 0x04563467"

I can print other files, even larger files, but this one will not load and print. 

Any ideas?  

Thanks, 
Chuck

chuck767, 2015-12-29:
Running Win7 Professional

chuck767, 2015-12-31:
anyone have any ideas what's going on here?????

dlarue, 2015-12-31:
you might try proving it's a problem in the current version to get some eyes on it. Once proven to exist in the current codebase you could then back port it to your 2014 version if there's a reason you are still on the old version.

chuck767, 2015-12-31:
Thanks, I will get the current version. I thought I had the current version.

chuck767, 2015-12-31:
The version I'm using came from this download

Printrun-Win-Slic3r-03Feb2015.zip

With this version loaded I see this "(Printrun 2014.08.01)" on the "abouit" tab.

Is there a newer version?

dlarue, 2015-12-31:
While I don't run Windows, I recalled seeing something about Printrun and Windows versions. Sure enough, see this issue: https://github.com/kliment/Printrun/issues/697

I don't know if the installations are in binary form or if they are just packaged source files. If there are Python source files, look into pronterface.py and printcore.py to find a 2015 date.

chuck767, 2015-12-31:
Thanks a lot, I will see what I can find. .

chuck767, 2015-12-31:
Being a newbie with python,  not sure how to go about checking the .py files. 

 The strange thing with this windows problem, is I have printed a lot of parts without any issues but this one particular part does not want to print. 
Two other people have loaded the file, one in  Pronterface  and one in Repetier. 

 I'm currently building a virtual machine to run the Linux version of Pronterface.  I would rather not do this but I can't seem to get the Windows version to work properly

dlarue, 2015-12-31:
It might be tedious but you could pin down the offending GCode line by making a copy of it and then edit and remove half of it and try loading the top half. If it works, start adding half of the bottom half back and try loading again. Eventually you should find the offending line and that might help better pinpoint the issue.  Or it's an out of memory error but most likely it's the 3D rendering so disabling preview or something to disable 3D rendering might work too.

chuck767, 2015-12-31:
Thanks, I will give that a try.

I actually looked at the gcode. the error  Said line 234 was the issue and
I didn't notice anything strange. Maybe it is a memory problem with too
much rendering. I need to look at it closer.

rockstorm101, 2017-12-14:
Hi, @chuck767, what is the status of this issue? Did you find what was wrong?

rockstorm101, 2018-02-24:
I'm closing this issue due to a lack of feedback. Please feel free to reopen if needed.

chuck767, 2018-07-10:
Please re-opened this issue. I never did solve my initial problem but I’m trying to print another part and I’m getting the same error and wondering if anyone else has any new ideas.
Regards,
Chuck

rockstorm101, 2018-07-10:
Hi, @chuck767,

 * What version of Printrun are you using? Have you tried the [latest one](https://github.com/kliment/Printrun/releases/tag/printrun-2.0.0rc5)?

 * Could you please share the file that is giving you problems? Just to see if we can reproduce your error.

Thanks

chuck767, 2018-07-11:
I'm using Printrun verson 2014.08.01

Here is the code I'm having problems with.

[Neck_with_supports21.txt](https://github.com/kliment/Printrun/files/2182836/Neck_with_supports21.txt)

chuck767, 2018-07-17:
I loader the latest version of printrun and all seems to be working fine now. 

 Thanks for your support Rock. 

Chuck

rockstorm101, 2018-07-17:
Good news. I'm sorry I couldn't do more. I'm "reclosing" then since it seems fixed.

closed 2018-07-17 by rockstorm101

chuck767, 2018-07-17:
> -sent from my iphone --

Rock,
You suggested I check for the latest version which I did and that was the problem. So you did great.

Chuck