kliment/Printrun #713: USB connection to Printrun Pronterface from El Capitan

mariyo66, 2016-01-06:
I just installed OS X El Capitan 10.11.2 and there is a unexpected problem: no USB connection to Printrun. I've changed to 3Feb2015 version and the problem persists. My Macbook Pro has two USB connections, but the problem is the same in both.
When I do the connection it appears 'tty.usbmodem1A21' port but now the process doesn't run and at the end I have to kill it to avoid wait for nothing.
For other things - connection to a traditional printer or another Apple device- there is no problem.
Any idea?
Thanks.
Mario

kpgresk, 2016-01-07:
I’m using a Mac Pro with OS 10.11.1 and can make the connection. Plug your printer into the USB port and the check Apple Menu > About thjs Mac > Overview > system report > Hardware > USB. Check the USB Device Tree for your connection. Select that connection in Printrun PORT.

Maybe this will help.

mariyo66, 2016-01-10:
Thanks for your info. I have just reinstalled El Capitan from zero to clean up possible problems. The result is the same. More over, I am suffering delays in the pointer and also in the keyboard. Sometimes it looses characters... this remembers me worst times of windows...

mariyo66, 2016-01-12:
Final solution: returning to Yosemite. To reach, look for the OS out of Apple environment, reinstall twice, using Time Machine, fix several problems, meawhile praying...

rdlaner, 2016-09-18:
Any luck with this issue?

rockstorm101, 2017-12-14:
Hi, @mariyo66, what's the status of this issue? Could we consider it closed since there's both new versions of Printrun and Mac OS?

Thanks

rockstorm101, 2018-03-06:
I'm closing this issue due to a lack of feedback. Please feel free to reopen if needed.

closed 2018-03-06 by rockstorm101

AllenZ-E5, 2020-12-02:
I am running Yosemite 10.10.5.  I don’t know what to put into the port box.

Thanks
Allen

AllenZ-E5, 2020-12-02:
Hi again
Someone above said to check USB Device Tree in System Report.

I don't see anything that looks like what I should use for the port in my MBP.  Can someone help please?
[image]

kliment, 2020-12-02:
It's the USB2.0-Serial

AllenZ-E5, 2020-12-02:
Maybe the baud rate i wrong then.  Still getting an error.
[image]

volconst, 2020-12-02:
@AllenZ-E5 , you probably have to install a driver for the serial chip on the 3d printer control board. 
Not a mac user, but I believe the port should be a file in /dev/* as in linux.
See https://stackoverflow.com/questions/17058134/is-there-an-equivalent-of-lsusb-for-os-x
Try to highlight the 'USB2.0-Serial' entry in System report. Does the lower pane change text?
If yes, search the VendorId number in google to identify the driver you need

AllenZ-E5, 2020-12-02:
Thanks.  There was 2 drivers available in the package.  Maybe I installed
the wrong one.  Totally forgot about that.
I'll install the other one.
Thanks

AllenZ-E5, 2020-12-03:
Driver is the right one and I opened it.  Should this be a terminal command
because Pronterface still won't connect.  Photo shows the error as well as
the terminal window.

One point of interest is when I plug in the USB cable the LCD screen on the
Ender 5 does light up.

Hopefully we can figure this out or I'll just go buy a cheap Chromebook
and there.s a controller that google has hidden
https://youtu.be/lR5QjnohsRk

Thanks
Allen

AllenZ-E5, 2020-12-03:
Sorry forgot the phorto

AllenZ-E5, 2020-12-05:
I'm in!  Through S3D.  got a USB extension cord and plugged into the
printer and over to my iMac.

Now I see a port but not sure if that will work on the MBP.  Gonna try that
next

Allen

rockstorm101, 2020-12-12:
Hi @AllenZ-E5, were you able to connect to your Ender 5? Could you please share with the community what the problem was? Was it the USB cable? A missing driver? Thanks

AllenZ-E5, 2020-12-12:
Hi.
Not through my MacBook Pro which is old.  Early 2008.  I was able to
connect with a long USB cable from my iMac through my S3D slicer.  There's
a hidden way to connect to a 3D printer through an app in the Chrome
browser.

Allen

AllenZ-E5, 2020-12-12:
Could never find a port to connect with my MBP.  So I gave up.  I had another problem I had to solve with the printer concerning there Z-Offset or the leveling sensor.  Finally got that solved.

No need for a usb connection now.