kliment/Printrun #716: how to export macros and share them ?

paboman, 2016-02-05:
Hi,
where the macros are saved?
are we able to share them between users?
regards
Patryk

dlarue, 2016-02-05:
The last I saw they were located within your printrunrc file.  To "export"
or share I just copied and pasted the macro(s) I was interested in sharing.

paboman, 2016-02-05:
thanks for the fast reply.
I can't find this file inside the "Printrun-Win-Slic3r-03Feb2015" folder

dlarue, 2016-02-05:
search for where to find the config file in Microsoft Windows. I know on
Linux it's $HOME/.pronsolerc

kliment, 2016-02-05:
It's usually in C:\Users\(username)\Appdata or something like that.
Depends on the windows version

paboman, 2016-02-05:
great!! under win is in C:\Users\(username)
in printrunconf.ini
thank you!!

hg42, 2016-04-21:
btw. what about examples?
Perhaps a folder somewhere could be filled with those?
May be simply share the bare config files? Everyone could take what he needs. This would also clarify the context of the macros/buttons.

E.g. I currently have about 50 buttons and 15 macros to share, using a Delta printer running on smoothieware.

Only very simple things, but useful every day: colored buttons, moving to certain positions/corners/center/towers/heights, extruder movement in two speeds, show position, set origin, touch the surface at certain heights for leveling support, bed and hotend temps, config show/load/save, reset, reboot, toggle debug, etc. The macros are used to make the buttons easier to write and to read.

any interest? should I create a pull request?

hg42, 2016-04-21:
[image]

soosalameh, 2017-02-02:
Hey guys, this thread is super helpful.
I was able to find the printrunconf file that had my macros.
I have one question though: is there a way to edit this file before having someone download the software, instead of having people copy and paste?