kliment/Printrun #719: Printer resets while bed and extruder heating

Antonstr, 2016-02-08:
Good day to you

The printer seems to reset once the bed ha reached 115 degrees and the extruder heats up. HAve to click on resume printing for it to start printing

rockstorm101, 2017-12-14:
Hi, @Antonstr,  what OS are you using? does this keep happening with the latest release?

Antonstr, 2017-12-15:
Good day I am using an Arduino Mega

rockstorm101, 2017-12-15:
Alright,

 * Does your printer keep resetting at 115 degrees?
 * What firmware do you use? Marlin?
 * Does this happen with Printrun 1.6.0?
 * What OS do you use? Windows?

Thanks

Antonstr, 2017-12-15:
Hi there

No it was random, upgraded to ver 1.0.2-2 seems to have sorted it out, I
put the printer one side for almost a year as I was busy developing a new
GPS system. Now working printing cases for the GPS
I am using Marlin yes 1.0.2-2)
Repetier i was using
Linux

Regards

Anton

kliment, 2017-12-15:
Good to hear it's resolved, thanks for updating us.

closed 2017-12-15 by kliment