kliment/Printrun #721: Cannot assign requested address (RPC?)

orangeudav, 2016-02-14:
```
DISPLAY=:1 python pronterface.py
WARNING:root:Memory-efficient GCoder implementation unavailable: No module named gcoder_line
WARNING:root:Could not setup DBus for sleep inhibition: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.ScreenSaver was not provided by any .service files
Xlib:  extension "RANDR" missing on display ":1".
Traceback (most recent call last):
  File "pronterface.py", line 35, in <module>
    app = PronterApp(False)
  File "/home/cubie/Printrun2/printrun/pronterface.py", line 2198, in __init__
    self.mainwindow = PronterWindow(self)
  File "/home/cubie/Printrun2/printrun/pronterface.py", line 173, in __init__
    self.parse_cmdline(sys.argv[1:])
  File "/home/cubie/Printrun2/printrun/pronsole.py", line 711, in parse_cmdline
    self.process_cmdline_arguments(args)
  File "/home/cubie/Printrun2/printrun/pronterface.py", line 884, in process_cmdline_arguments
    pronsole.pronsole.process_cmdline_arguments(self, args)
  File "/home/cubie/Printrun2/printrun/pronsole.py", line 698, in process_cmdline_arguments
    self.update_rpc_server(None, self.settings.rpc_server)
  File "/home/cubie/Printrun2/printrun/pronsole.py", line 670, in update_rpc_server
    self.rpc_server = ProntRPC(self)
  File "/home/cubie/Printrun2/printrun/rpc.py", line 37, in __init__
    logRequests = False)
  File "/usr/lib/python2.7/SimpleXMLRPCServer.py", line 593, in __init__
    SocketServer.TCPServer.__init__(self, addr, requestHandler, bind_and_activate)
  File "/usr/lib/python2.7/SocketServer.py", line 419, in __init__
    self.server_bind()
  File "/usr/lib/python2.7/SocketServer.py", line 430, in server_bind
    self.socket.bind(self.server_address)
  File "/usr/lib/python2.7/socket.py", line 224, in meth
    return getattr(self._sock,name)(*args)
socket.error: [Errno 99] Cannot assign requested address
```

```
netstat -lntp
(Not all processes could be identified, non-owned process info
 will not be shown, you would have to be root to see it all.)
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name
tcp        0      0 0.0.0.0:36000           0.0.0.0:*               LISTEN      -
tcp        0      0 0.0.0.0:5901            0.0.0.0:*               LISTEN      31716/Xtightvnc
tcp        0      0 0.0.0.0:111             0.0.0.0:*               LISTEN      -
tcp        0      0 0.0.0.0:6001            0.0.0.0:*               LISTEN      31716/Xtightvnc
tcp        0      0 0.0.0.0:49523           0.0.0.0:*               LISTEN      -
tcp6       0      0 :::36000                :::*                    LISTEN      -
tcp6       0      0 :::60682                :::*                    LISTEN      -
tcp6       0      0 :::111                  :::*                    LISTEN      -
```

orangeudav, 2016-02-14:
Temporally disable this shit
```
diff --git a/printrun/pronsole.py b/printrun/pronsole.py
index e449457..7409478 100644
--- a/printrun/pronsole.py
+++ b/printrun/pronsole.py
@@ -667,7 +667,7 @@ class pronsole(cmd.Cmd):
     def update_rpc_server(self, param, value):
         if value:
             if self.rpc_server is None:
-                self.rpc_server = ProntRPC(self)
+                self.rpc_server = None  #ProntRPC(self)
         else:
             if self.rpc_server is not None:
                 self.rpc_server.shutdown()
```

oli44, 2016-03-24:
it works!

rockstorm101, 2017-12-14:
Hi, @a0s, does this keep happening with the latest release? What OS are you using?

rockstorm101, 2018-03-06:
I'm closing this issue due to a lack of feedback. Please feel free to reopen if needed.

closed 2018-03-06 by rockstorm101

u371, 2018-12-08:
I don't really want to re-open this but add some pointers/comments.
I found this page when I ran into the same error recently. For the last year or so I had been forced to start pronterface as the very first program after booting the computer or get this same error. Then a recent kernel update completely blocked starting the program. Since there are few alternatives available, I wanted to get it going again. My version is older and the above code isn't available, so I couldn't use this fix but it got me thinking.
grep -R -i rpc_server in the printrun directory got me the following response:
printrun/settings.py:        self._add(BooleanSetting("rpc_server", True, _("RPC server"), _("Enable RPC server to allow remotely querying print status")), root.update_rpc_server)
Changing 'True' to 'False' resolved all my problems.
My only question would be from a design point of view: does this server really need to be turned on by default? Communications between my computer and printer via bluetooth run fine without it.