kliment/Printrun #723: M105 is not sent with linenumber / checksum

AnHardt, 2016-02-22:
Even when running a g-code program Pronterface is sending M105 without line number and checksum.
Is that done by purpose, to prevent a special firmware from misinterpreting?
If not, this prevents firmwares from checking the protocol a bit more strict.

kliment, 2016-02-22:
It's being done for implementation reasons - the M105s that are being
sent with the "monitor printer status" checkbox selected are skipping
the queue of numbered commands and being sent as soon as possible. The
same is the case for commands typed into the console. The no-checksum
M105 is only really needed during initial communication because some
printers reset on connect and a number of initial messages can be lost.
During this time, the host sends M105 in a loop every few seconds until
a response comes back. For the M105s sent after that, during prints,
it's possible to enable checksums and line numbers, but this is not
currently enabled. If you want to enable it, the relevant piece of code
is in the _sendnext(self) function in printrun/printcore.py. To add
checksums and line numbers, in that function, change the line

self._send(self.priqueue.get_nowait())

to

self._send(self.priqueue.get_nowait(), self.lineno, True)

If you want to avoid the M105s altogether, you can disable "monitor
printer status" in the settings.

AnHardt, 2016-02-22:
Thanks a lot, for the explanation.

closed 2016-02-22 by AnHardt