kliment/Printrun #725: Uncertain or missing licesing information

rockstorm101, 2016-02-28:
Hi, I would like to know if this software if fully distributable under the GPL-3 license.

For the files contained in locale/, applicable license is not specified, and in the following two files it is not stated neither a copyright holder nor an applicable license, does GPL-3 truly apply to them?

    printrun/gcoder_line_extra.h
    setup_win.py
    
Thanks a lot,
Rock

iXce, 2016-02-28:
I'm fairly confident about printrun/gcoder_line_extra.h which I wrote, except the typedef, which comes from guppy.
However, to the best of my knowledge, this file is not used by default (you have to manually apply gcoder_heapy_support.patch	and recompile the pyrex/cython file) and could be either dropped or kept out of the standard distribution.

kliment, 2016-02-28:
I'm the author of setup_win.py and I believe it's not copyrightable in itself due to triviality, but if it's copyrightable, GPLv3 applies

rockstorm101, 2016-03-01:
Thank you guys, please forgive me but I must insist on this matter even though you are not the authors of most of them, what about all the files in the "locale" folder? could we say that the GPL v3 applies to them too?

    locale/*

Typically, a file stating a copyright holder but no license is assumed to be an "all rights reserved" file.

Thanks again for your kind help.

kliment, 2016-03-01:
All those files are derivatives of GPLv3 licensed source files, so GPLv3
applies to them.

rockstorm101, 2016-03-13:
Thanks a lot, I consider this matter closed.

closed 2016-03-13 by rockstorm101