kliment/Printrun #729: Debian Jessie (armhf) and wxPython

Arthedian, 2016-03-21:
Hi,
I tried to use this package on my Beaglebone black with Debian Jessie, but it ran with this error "E: Package 'python-wxgtk2.8' has no installation candidate". Does anyone know how to fix this? I tried to google it, but I couldn't find anything.

rockstorm101, 2016-03-21:
Hi @Arthedian ,

The current version of the Python interface to the wxWidgets on Debian Jessie is 3.0, you must install the package "python-wxgtk3.0" instead of "python-wxgtk2.8".

pattex16, 2019-01-04:
> The current version of the Python interface to the wxWidgets on Debian Jessie is 3.0, you must install the package "python-wxgtk3.0" instead of "python-wxgtk2.8".

Hey, i'm using Debian Jessie on armhf as well but despite installing python-wxgtk3.0, i still cant import wx and if i check my installed pip packages with `pip list`, wx doesnt show up. What can i do to make it work?  `sudo pip3 install wxPython` leads to a setup error btw.

rockstorm101, 2019-01-18:
> Hey, i'm using Debian Jessie on armhf as well but despite installing python-wxgtk3.0, i still cant import wx and if i check my installed pip packages with pip list, wx doesnt show up. What can i do to make it work? sudo pip3 install wxPython leads to a setup error btw.

@pattex16, are you following the steps stated in the README [[1]]? As is explained, "wxPython4 doesn't have Linux wheels available", therefore `pip install wxPython` won't work.

Particularly for Debian Jessie, for installing wxPython you should:
```
python -m pip install https://extras.wxpython.org/wxPython4/extras/linux/gtk3/debian-8/wxPython-4.0.4-cp34-cp34m-linux_x86_64.whl`
```
Next time, I believe it is better if you either reopen or create a new bug, instead of  posting on closed bugs. If the given solution works, please close this bug again.

[1]: https://github.com/kliment/Printrun#use-python-virtual-environment

rockstorm101, 2020-05-23:
Due to the lack of feedback I assume the given solution worked. Please feel free to reopen if needed.

PR #730 (2016-05-28): Update installation in Ubuntu/Debian info
 * Added information regarding the new package for Debian Sid.
 * Updated dependency for Ubuntu/Debian systems "python-wxgtk2.8" to
   version 3.0. ( Closes #729 )
closed 2020-05-23 by rockstorm101