kliment/Printrun #733: Error and spurious G-code sent at end of print

kognix, 2016-04-04:
At the end of a print Pronterface is sending unwanted G-code, particularly a G1 Z100 followed by a blank line that causes a runtime error:
[ERROR] Traceback (most recent call last):
  File "/home/ian/apps/Printrun-printrun-20150310/printrun/printcore.py", line 605, in _send
    try: self.sendcb(command, gline)
  File "/home/ian/apps/Printrun-printrun-20150310/printrun/pronterface.py", line 1558, in sentcb
    if gline.is_move:
AttributeError: 'NoneType' object has no attribute 'is_move'

The Z-axis move is a serious problem! On my printer (own design running Marlin) it either crushes the print or physically crashes the extruder depending on the XY location at end of job. Below is the communications debug output of a very simple print. As can be seen this error occurs in the 20150310 release but also in a current master clone.

SENT: N162 G28 X0*94
RECV: ok
SENT: M105
RECV: ok T:181.7 /0.0 B:59.3 /0.0 @:0 B@:0
SENT: N163 M84*27
RECV: ok
SENT: N-1 M110*15
Print ended at: 19:27:49 and took 0:00:26
RECV: ok
SENT: M104 T1 S40.
SENT: G1 Z100.0 F80.
SENT: 
RECV: echo:M104 Invalid extruder 
RECV: ok
RECV: ok
[ERROR] Traceback (most recent call last):
  File "/home/ian/apps/Printrun-printrun-20150310/printrun/printcore.py", line 605, in _send
    try: self.sendcb(command, gline)
  File "/home/ian/apps/Printrun-printrun-20150310/printrun/pronterface.py", line 1558, in sentcb
    if gline.is_move:
AttributeError: 'NoneType' object has no attribute 'is_move'

kliment, 2016-04-04:
Wow, that is weird. I don't think I've ever seen it happen before. Can I
see the gcode file that causes it.

kognix, 2016-04-04:
Thanks for the quick response! The gcode is attached - just a simple tiny cube for debugging this issue, but it happens with all my prints. Sliced with skeinforge, with my own end of print homing commands.

[524_export.gcode.txt](https://github.com/kliment/Printrun/files/203356/524_export.gcode.txt)

kliment, 2016-04-04:
Honestly I have no idea what's going on. Do you have a final command in
your pronterface options? Does it happen with the current master off github?

kognix, 2016-04-04:
There are no start or final commands in my Pronterface settings - I set those in my slicer. Same problem with current master. Only discovered this when configuring a newer host to control the printer - an old version of Pronterface (from 2013) didn't exhibit the problem.
Trying to debug this, but I'm no Python expert. Wondering about the Z moves in projectlayer.rise ? maybe...

kognix, 2016-04-05:
Curious problem this. After poring through the code and debugging it's clear the spurious gcode isn't being issued directly from pronterface. I've never set any final command, it was blank in the external commands dialog and I double checked in .pronsolerc that there was no final_command. However, having set it to "unicorn.txt" (it doesn't exist) through the interface, the next test was OK.
Of course it threw a "no such file" error, but the spurious gcode wasn't transmitted and my end of print was as expected:

SENT: N162 G28 X0*94
RECV: ok
SENT: M105
RECV: ok T:182.0 /0.0 B:59.3 /0.0 @:0 B@:0
SENT: N163 M84*27
RECV: ok
SENT: N-1 M110*15
RECV: ok
Print ended at: 22:49:09 and took 0:00:28
[ERROR] Print end callback failed with:
Traceback (most recent call last):
  File "/home/ian/apps/Printrun-printrun-20150310/printrun/printcore.py", line 503, in _print
    try: self.endcb()
  File "/home/ian/apps/Printrun-printrun-20150310/printrun/pronterface.py", line 1511, in endcb
    pronsole.pronsole.endcb(self)
  File "/home/ian/apps/Printrun-printrun-20150310/printrun/pronsole.py", line 1197, in endcb
    "$t": format_duration(print_duration)})
  File "/home/ian/apps/Printrun-printrun-20150310/printrun/utils.py", line 139, in get_command_output
    blocking = False)
  File "/home/ian/apps/Printrun-printrun-20150310/printrun/utils.py", line 134, in run_command
    return subprocess.Popen(command, stderr = stderr, stdout = stdout)
  File "/usr/lib/python2.7/subprocess.py", line 710, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1327, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory

SENT: M105
RECV: ok T:179.8 /0.0 B:58.7 /0.0 @:0 B@:0

kognix, 2016-04-05:
And now after a long grep I find a file "end.gcode" with the following contents, i.e. the spurious commands that were being sent at end of print:
M104 T1 S40.0
G1 Z100.0 F80.0
This file is in an entirely separate directory from Pronterface. Wondering how and why it was being processed?!

kliment, 2016-04-06:
Good detective work! This is contributed code so I didn't spot it at
first. If a file called end.gcode or pause.gcode is found in the current
working directory, it will be run on print end/pause. Possibly because
it's in your working directory it was able to find it. I'll set the
default to None rather than end.gcode so noone else gets accidentally
bitten by this.

closed 2016-04-06 by kliment

kognix, 2016-04-06:
Many thanks! Much appreciated.

commit 0193a9db (2016-04-06): Disable parsing of end.gcode and pause.gcode by default so an unintended end.gcode in the working directory doesn't perform unexpected commands. Fixes #733