kliment/Printrun #738: Provide signed DMG package for Mac

larsrc, 2016-04-14:
I was happily using Printrun on Linux until recently, where I switched to Mac. Due to company policy, I cannot install unsigned DMG packages on the Mac. Could you provide a signed DMG? Instructions are here: https://developer.apple.com/library/mac/documentation/IDEs/Conceptual/AppDistributionGuide/DistributingApplicationsOutside/DistributingApplicationsOutside.html

kliment, 2016-04-14:
Hi! Unfortunately I don't have a developer ID or a mac modern enough to run any xcode version with signing capability. If you have one, would you be willing to build mac binaries?

ntoff, 2016-04-24:
How do you build the mac binaries? Same way in windows with that batch file?

edyweber, 2016-11-22:
kliment... PrintRun stop running in my mac... No reason.. Yesterday was working ok... It does not open (app no responding). Any help?

hroncok, 2018-03-26:
@syky27 is my colleague from @3DprintFIT. He has the account and stuff needed to sign the macOS binaries.

@kliment Could you please instruct him on what exactly is needed?

hroncok, 2018-03-26:
How to build is at https://github.com/kliment/Printrun/blob/2.x/buildinstructions.txt

kliment, 2018-03-26:
There is an unsigned binary at https://github.com/kliment/Printrun/releases/download/printrun-2.0.0rc5/pronterface-mac-2.0.0rc5.zip right now - you'd need to unpack it, and then run

``codesign -s identityname pronterface.app --deep

Where identityname is your apple/signingkey ID.
I don't know what is needed to then package it into a dmg, but once that is done I believe the dmg needs to be signed the same way (without --deep iirc).

kliment, 2018-03-26:
Alternatively you can follow the build instructions linked above locally and build it yourself - however in that case you need to edit the dist/pronterface.app/Contents/Info.plist and edit the version (set to 2.0.0rc5) and add the NSAppSleepDisabled key.

kliment, 2018-03-26:
Here is how to create a dmg:
- copy the .app into a new folder.
- open Disk Utility -> File -> New Image -> Image From Folder.
- select the folder with the .app, give the DMG a name and save.

syky27, 2018-03-26:
@kliment Hi, I'll do that tomorrow, sorry for the late response...

kliment, 2018-03-26:
No hurry at all, @syky27 and thank you so much!

syky27, 2018-03-28:
I have created symlink to `/Applications` to the .dmg, for better usability.

The app itself is signed as well as the .dmg

Please test when you get a chance.

Sorry for the link but the archive is too big to upload to GH.

https://www.dropbox.com/s/9pl1amezeycepfc/Signed_Pronterface.zip?dl=0

kliment, 2018-03-28:
Thanks, looks great, I've uploaded it to the release page.

kliment, 2018-03-28:
Awesome, thank you so much! @FidelCapo, can you test the DMG please?

FidelCapo, 2018-03-29:
@kliment When I downloaded pronterface from the link that provided @syky27 , dmg and the app looks signed and I can open them. The Pronterface I downloaded from github relases (https://github.com/kliment/Printrun/releases/download/printrun-2.0.0rc5/pronterface-mac-2.0.0rc5.zip) is not signed.

kliment, 2018-03-29:
@FidelCapo that is correct - I've uploaded the signed dmg to the releases also now. The .zip is still unsigned.

FidelCapo, 2018-03-29:
@kliment ok

closed 2018-09-12 by kliment

rockstorm101, 2020-07-03:
Dear @hroncok and @syky27, I believe we are very close to being able to release (at last) version 2.0.0 of Printrun. We face the same inconvenience as last time, we lack "developer ID or a mac modern enough to run any xcode version with signing capability". Could you please help again on this regard when the time comes?

syky27, 2020-07-03:
@rockstorm101  sure thing, let me know...