kliment/Printrun #739: error in log file

KevinBWood, 2016-04-21:
WARNING:root:Memory-efficient GCoder implementation unavailable: No module named gcoder_line
F:\0E5B-C145\WQ0003资料包V5.3\软件源码\Printrun-master\Printrun-master\dist\library.zip\pyglet\window\win32\__init__.py:552: UserWarning: Could not set vsync; unsupported extension.

billy0329, 2016-04-26:
Check this
[https://github.com/kliment/Printrun/blob/master/README.md#cython-based-g-code-parser](url)

to remove this warning:

WARNING:root:Memory-efficient GCoder implementation unavailable: No module named gcoder_line

as for the second one, i have no idea. Maybe you can provide more details.

rockstorm101, 2017-12-14:
Hi, @KevinBBWood, do you keep seeing this errors with the latest release?

billy0329, 2017-12-14:
I think the issue here is that because gcoder_line.pyd is not present. It must be compiled from gcoder_line.pyx. The instruction is in README.md

rockstorm101, 2017-12-15:
I agree. First line is surely produced because gcoder_line is not present. But the second one seems related to pyglet.

closed 2018-02-24 by rockstorm101