kliment/Printrun #743: Steps in readme not enough to run pronterface on Windows

iHateInventNames, 2016-04-30:


VanessaE, 2016-04-30:
Could you be more specific about this issue?

What version or commit ID are you trying to use?
What version of Windows is this?
What exactly happens when Printrun fails?

Tinchus2009, 2016-05-04:
I have the same problem, I have win 10, I followed the instructions posted here, downloaded all files, installed them in the specified order. All of them intalled correctly.
Then I go to the folder where I unpacked the .zip file I downloaded from here, containing all pronterface files. I double click on pronterface.py and it fails, this is what I get in a console messaage:

WARNING:root:Memory-efficient GCoder implementation unavailable: No module named gcoder_line
WARNING:root:psutil unavailable, could not import power utils:No module named psutil
[ERROR] 3D view mode requested, but we failed to initialize it.
Falling back to 2D view, and here is the backtrace:
Traceback (most recent call last):
  File "C:\Users\Martin\Downloads\instalacion pronterface\Printrun-master\Printrun-master\printrun\gui\viz.py", line 70, in __init__
    import printrun.gcview
  File "C:\Users\Martin\Downloads\instalacion pronterface\Printrun-master\Printrun-master\printrun\gcview.py", line 22, in <module>
    from .gl.panel import wxGLPanel
  File "C:\Users\Martin\Downloads\instalacion pronterface\Printrun-master\Printrun-master\printrun\gl\panel.py", line 21, in <module>
    import numpy
ImportError: No module named numpy

Traceback (most recent call last):
  File "C:\Users\Martin\Downloads\instalacion pronterface\Printrun-master\Printrun-master\pronterface.py", line 59, in <module>
    app = PronterApp(False)
  File "C:\Users\Martin\Downloads\instalacion pronterface\Printrun-master\Printrun-master\printrun\pronterface.py", line 2216, in __init__
    self.mainwindow = PronterWindow(self)
  File "C:\Users\Martin\Downloads\instalacion pronterface\Printrun-master\Printrun-master\printrun\pronterface.py", line 218, in __init__
    self.reload_ui()
  File "C:\Users\Martin\Downloads\instalacion pronterface\Printrun-master\Printrun-master\printrun\pronterface.py", line 280, in reload_ui
    self.settings.controlsmode == "Mini")
  File "C:\Users\Martin\Downloads\instalacion pronterface\Printrun-master\Printrun-master\printrun\gui\__init__.py", line 256, in createGui
    viz_pane = VizPane(self, vizpanel)
  File "C:\Users\Martin\Downloads\instalacion pronterface\Printrun-master\Printrun-master\printrun\gui\viz.py", line 79, in __init__
    from printrun import gviz
  File "C:\Users\Martin\Downloads\instalacion pronterface\Printrun-master\Printrun-master\printrun\gviz.py", line 18, in <module>
    import numpy
ImportError: No module named numpy

kliment, 2016-05-04:
Okay, I think I know what's happening - numpy is needed for the 3d view to work, but it's not included in the installation instructions. Can you check if installing numpy fixes the problem? If it does, I'll update the instructions to reflect that.

Tinchus2009, 2016-05-04:
Ok. I did it, you need to install numpy. From source is a nightmare of dependencies, so I found this binaries for windows (I used the 32 bit version, despite I have win 10 64, but the 64 bits version didnt worked): http://www.lfd.uci.edu/~gohlke/pythonlibs/
To install that package you need to uso PIP, wich is already installed in python (probably you will need to use it finding the full path )
I run pronterface.py , and it started. Will later try if it actually works, but at least it started. In the console I get this warning:

WARNING:root:Memory-efficient GCoder implementation unavailable: No module named gcoder_line
WARNING:root:psutil unavailable, could not import power utils:No module named psutil

iHateInventNames, 2016-05-04:
also wxPython-phoenix is only available through pip for Windows

rockstorm101, 2016-12-08:
Hi @iHateInventNames and @Tinchus2009, any updates on this? Did you finally manage to compile printrun on Windows?

pcarew, 2017-02-28:
Problem running/starting Pronterface.py

I have installed the packages listed in the Readme, but when I try and start Pronterface, I get two warnings and then nothing..
Here is what I receive:
c:\Users\Carew\Downloads\Printrun-master\Printrun-master>python pronterface.py
WARNING:root:Memory-efficient GCoder implementation unavailable: No module named gcoder_line
WARNING:root:psutil unavailable, could not import power utils:No module named psutil

At this point nothing else happens.

apphead, 2017-09-16:
Just wasted a loooot of time achieving nothing following the "run from source" instructions for windows (8x64). As there has been no release for 2+ years for windows, I think it would be appreciated by the community to at least provide a working set of requirements and instructions!