kliment/Printrun #745: G28 Zeroing Question

chevellebro1, 2016-05-09:
I would like to run the homing sequence without zeroing the machine. I can't find in the code where there is a command to zero when an endstop is hit. Can someone shed some light on this?

```
#define HOMEAXIS_DO_LIDAR(LETTER) \
  ((LETTER##_MIN_PIN > -1 && LETTER##_HOME_DIR==-1) || (LETTER##_MAX_PIN > -1 && LETTER##_HOME_DIR==1))

  if (axis==X_AXIS ? HOMEAXIS_DO_LIDAR(X) :
      axis==Y_AXIS ? HOMEAXIS_DO_LIDAR(Y) :
      axis==Z_AXIS ? HOMEAXIS_DO_LIDAR(Z) :
      0) {

    int axis_home_dir = home_dir(axis) * -1;

    current_position[axis] = 0;
    plan_set_position(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS], current_position[E_AXIS]);

    destination[axis] = 1.5 * max_length(axis) * axis_home_dir;
    feedrate = 500;

    plan_buffer_line(destination[X_AXIS], destination[Y_AXIS], destination[Z_AXIS], destination[E_AXIS], feedrate/60, active_extruder);
    st_synchronize();

    current_position[axis] = 0;
    plan_set_position(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS], current_position[E_AXIS]);
    destination[axis] = -home_retract_mm(axis) * axis_home_dir;
    plan_buffer_line(destination[X_AXIS], destination[Y_AXIS], destination[Z_AXIS], destination[E_AXIS], feedrate/60, active_extruder);
    st_synchronize();

    destination[axis] = 2*home_retract_mm(X_AXIS) * axis_home_dir;
    feedrate = 100;

    plan_buffer_line(destination[X_AXIS], destination[Y_AXIS], destination[Z_AXIS], destination[E_AXIS], feedrate/60, active_extruder);
    st_synchronize();

    axis_is_at_home(axis);
    destination[axis] = current_position[axis];
    feedrate = 0.0;
    endstops_hit_on_purpose();
    axis_known_position[axis] = true;

    }
}

kliment, 2016-05-09:
This is really not the appropriate place to ask - it's a question for the firmware's repository.
However, I understand the internals of Marlin to some degree and will try to help - to do what you want to do, save the destination and current position values for each axis to some variables, and restore them after the homing sequence is run. 
The homing sequence sets a destination far beyond the endstop to ensure it will be hit regardless of position and so the machine loses its known position.
After the homing sequence is run, the G28 handler explicitly sets the current position and destination for the axis to the homing position.
So, if you want to avoid that, save the destination values to an array before you run your lidar command, and then set the destination and current position to those values after the homing sequence has run.

rockstorm101, 2017-12-14:
The question was given an answer and it was not printrun-related. Closing.

closed 2017-12-14 by rockstorm101