kliment/Printrun #749: Problems with OSX compiled Printrun

eagleapex, 2016-05-28:
I get these errors constantly in my console:
`[ERROR] Failed to set power settings:
Traceback (most recent call last):
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.pronsole", line 1173, in startcb
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.power", line 130, in powerset_print_start
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.power.osx", line 72, in inhibit_sleep_osx
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.power.osx", line 49, in AssertionCreateWithName
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.power.osx", line 40, in StringToCFString
AttributeError: 'module' object has no attribute 'kCFStringEncodingASCII'`

To my untrained eye it looks like "craigvanvliet" compiled something local that I don't have.

rockstorm101, 2016-05-29:
This is the same error reported in #619, and it is supposed to be fixed by commit e38998c, which came after the last precompiled Printrun for OSX was built. Could you try that workaround and see if it works?

eagleapex, 2016-06-01:
I don't have the resources to compile printrun and rely on the precompiled Printrun.

rockstorm101, 2017-12-14:
Hi, @eagleapex, is this finally fixed with the latest release?

eagleapex, 2017-12-14:
I will test this!

Nocturtle, 2018-02-13:
I am also having this issue with a mac pro (early 2009) running the latest release on OS X 10.11.6.

kliment, 2018-03-26:
@eagleapex  can you test this against he 2.0.0rc5 binary?

eagleapex, 2018-03-29:
I can't install from source. I was using the precompiled OSX binaries here:
http://koti.kapsi.fi/~kliment/printrun/

Chris Thompson, eagleapex.com
"Obscurity is a far greater threat to authors and creative artists than
piracy" -Tim O'Reilly

kliment, 2018-03-29:
@eagleapex there are fresh binaries at https://github.com/kliment/Printrun/releases

eagleapex, 2018-03-30:
2.0.0rc5 Works great on my Macbook and Prusa I3 MK3. Thanks!

Chris Thompson, eagleapex.com
"Obscurity is a far greater threat to authors and creative artists than
piracy" -Tim O'Reilly

rockstorm101, 2020-05-08:
Closing. Fixed by e38998c. Thanks.

closed 2020-05-08 by rockstorm101