kliment/Printrun #758: PyDeadObjectError: The C++ part of the PronterWindow object has been deleted

samspade4321, 2016-07-16:
I  get this error when closing Pronterface.  Occurs in both Win7 Pro and Windows 10 Pro.  Everything else seems to be OK.

Version: Printrun 2014.08.01

Log file:
Exception in thread Thread-12:
Traceback (most recent call last):
  File "threading.pyc", line 808, in __bootstrap_inner
  File "threading.pyc", line 761, in run
  File "printrun\pronterface.pyc", line 1006, in statuschecker
  File "printrun\pronsole.pyc", line 860, in statuschecker
  File "printrun\pronterface.pyc", line 998, in statuschecker_inner
  File "wx\_core.pyc", line 14619, in __getattr__
PyDeadObjectError: The C++ part of the PronterWindow object has been deleted, attribute access no longer allowed.

C:\WinSetupApp x64\Printrun\Printrun-Win-Slic3r-03Feb2015\library.zip\pyglet\window\win32\__init__.py:552: UserWarning: Could not set vsync; unsupported extension.

rockstorm101, 2017-12-14:
Hi, does this keep happening with the latest release?

Doug46465164654646, 2018-01-14:
I have been using Printrun for a long time, I have the current version so I thought; I am not sure now.

Anyway, I was using Printrun the last couple of days and all was good, then I decided to used sketchup and I didn't notice that the last update made it use the INCHES template and I created a stl file by exporting to mm... obviously the slicer took too long to slice it and I closed everything and discovered what I had done... So I deleted the slicer export file... and did another sketchup this time using the right 3d printer mm template... and opened Printrun and opened the tiny 48mmx28mmx6mm print and that is when I got the error message above and every time since. 

I am going to delete all my Printrun files on my windows 10 computer running advast free antivirus and try again. I am just added this information as it might be helpful to others. 

[image]

rockstorm101, 2018-03-06:
Hi guys, could you please check whether this keeps happening on the latest release?

kliment, 2018-03-27:
This was the race confition on exit bug, which has been fixed in 2.x and should no longer happen with the 2.0.0rc5 or later releases.

zhangli709, 2019-10-28:
少的地方

rockstorm101, 2020-05-23:
I believe the reported error was solved by b48fe7b. Please feel free to reopen if needed.

closed 2020-05-23 by rockstorm101