kliment/Printrun #760: Time estimate produced by printrun/gcoder.py varies

highfellow, 2016-08-11:
The time estimate produced by the gcode analyser `printrun/gcoder.py` varies between runs when I run it from the terminal against the same file. I compiled it with the efficient gcode analyser with the command: `python setup.py build_ext --inplace`. Below is a log of some of my results, using the same gcode file.

```
andy@monkey:~/Desktop/projects/red-hot-fountain-pen$ gcoder rhfp-midsection.gcode 
Line object size: 80
Light line object size: 40
Dimensions:
	X: 127.55 - 170.45 (42.89)
	Y: 114.59 - 160.43 (45.85)
	Z: 0.00 - 15.84 (15.84)
Filament used: 769.02mm
E0 769.02mm
Number of layers: 110
Estimated duration: 0:31:01
andy@monkey:~/Desktop/projects/red-hot-fountain-pen$ gcoder rhfp-midsection.gcode 
Line object size: 80
Light line object size: 40
Dimensions:
	X: 127.55 - 170.45 (42.89)
	Y: 114.59 - 160.43 (45.85)
	Z: 0.00 - 33.44 (33.44)
Filament used: 1574.69mm
E0 1574.69mm
Number of layers: 229
Estimated duration: 1:00:12
andy@monkey:~/Desktop/projects/red-hot-fountain-pen$ gcoder rhfp-midsection.gcode 
Line object size: 80
Light line object size: 40
Dimensions:
	X: 127.55 - 170.45 (42.89)
	Y: 114.59 - 160.43 (45.85)
	Z: 0.00 - 51.92 (51.92)
Filament used: 2425.24mm
E0 2425.24mm
Number of layers: 358
Estimated duration: 1:33:01
andy@monkey:~/Desktop/projects/red-hot-fountain-pen$ gcoder rhfp-midsection.gcode 
Line object size: 80
Light line object size: 40
Dimensions:
	X: 127.55 - 170.45 (42.89)
	Y: 114.59 - 160.43 (45.85)
	Z: 0.00 - 71.10 (71.10)
Filament used: 3305.61mm
E0 3305.61mm
Number of layers: 497
Estimated duration: 2:07:17
andy@monkey:~/Desktop/projects/red-hot-fountain-pen$ gcoder rhfp-midsection.gcode 
Line object size: 80
Light line object size: 40
Dimensions:
	X: 127.55 - 170.45 (42.89)
	Y: 114.59 - 160.43 (45.85)
	Z: 0.00 - 100.76 (100.76)
Filament used: 4628.16mm
E0 4628.16mm
Number of layers: 700
Estimated duration: 2:56:14
andy@monkey:~/Desktop/projects/red-hot-fountain-pen$ gcoder rhfp-midsection.gcode 
Line object size: 80
Light line object size: 40
Dimensions:
	X: 127.55 - 170.45 (42.89)
	Y: 114.59 - 160.43 (45.85)
	Z: 0.00 - 122.10 (122.10)
Filament used: 5620.45mm
E0 5620.45mm
Number of layers: 841
Estimated duration: 3:35:09
andy@monkey:~/Desktop/projects/red-hot-fountain-pen$ gcoder rhfp-midsection.gcode 
Line object size: 80
Light line object size: 40
Dimensions:
	X: 127.55 - 170.45 (42.89)
	Y: 114.59 - 160.43 (45.85)
	Z: 0.00 - 122.10 (122.10)
Filament used: 5620.45mm
E0 5620.45mm
Number of layers: 841
Estimated duration: 3:35:09
andy@monkey:~/Desktop/projects/red-hot-fountain-pen$ gcoder rhfp-midsection.gcode 
Line object size: 80
Light line object size: 40
Dimensions:
	X: 127.55 - 170.45 (42.89)
	Y: 114.59 - 160.43 (45.85)
	Z: 0.00 - 122.10 (122.10)
Filament used: 5620.45mm
E0 5620.45mm
Number of layers: 841
Estimated duration: 3:35:09

```

highfellow, 2016-08-11:
I've looked at this again and it's my mistake - I was running gcoder before Slic3r had finished writing the gcode.

rockstorm101, 2017-12-14:
The question seems solved. I'm closing the issue.

closed 2017-12-14 by rockstorm101