kliment/Printrun #761: Long macros not sent completely

VanessaE, 2016-08-15:
For particularly long macros, it seems only the first 15 or so lines are sent to the printer, or at least the code is not executed.  The rest seems to just get "lost" in the aether.  This one, for example, unloads the filament from my secondary extruder:
```gcode
G91
G1 Z20 F1000
G90
T1
G28 X
G92 E0
G1 E30F250
G92 E0
G1 E-100 F5000
G92 E0
G1 E-100 F5000
G92 E0
G1 E-100 F5000
G92 E0
G1 E-100 F5000
G92 E0
G4 S2
G1 E-100 F5000
G92 E0
G1 E-100 F5000
G92 E0
G1 E-100 F5000
G92 E0
G1 E-100 F5000
G92 E0
T0
G28 X
G1 F15000 X100
```

I realize this is probably not the best way to do things, but other than the fact that it doesn't make it all the way to that last G1 command, it otherwise works.  In fact, it seems to die after about 6 of those G1/G92 pairs, in the middle of one G1 lines, as it seems like a G1 E-100 does get executed, but at an extremely slow speed (as if it only caught "G1 E-100 F5" or something).

This similar macro for unloading my primary extruder makes it all the way:

```gcode
G91
G1 Z20 F1000
G90
G1 E-100 F5000
G92 E0
G1 E-100 F5000
G92 E0
G1 E-100 F5000
G92 E0
G4 S2
G1 E-100 F5000
G92 E0
G1 E-100 F5000
G92 E0
G1 E-100 F5000
G92 E0
G1 E-100 F5000
G92 E0
G1 E-100 F5000
G92 E0
```

kliment, 2018-03-08:
Can you check if this still happens with the 2.x version?

VanessaE, 2018-03-08:
It's still a problem at least for the manual command entry.  Paste this (a simple series of moves at max speed to distribute lube and check for axis binding):

```gcode
M201 T2000
g1 x0 f24000
g1 y0 f24000
g1 x200 f24000
g1 y200 f24000
g1 x0 f24000
g1 y0 f24000
g1 x200 f24000
g1 y200 f24000
g1 x0 f24000
g1 y0 f24000
g1 x200 f24000
g1 y200 f24000
g1 x0 f24000
g1 y0 f24000
g1 x200 f24000
g1 y200 f24000
g1 x0 f24000
g1 y0 f24000
g1 x200 f24000
g1 y200 f24000
g1 x0 f24000
g1 y0 f24000
g1 x200 f24000
g1 y200 f24000
g1 x0 f24000
g1 y0 f24000
g1 x200 f24000
g1 y200 f24000
g1 x0 f24000
g1 y0 f24000
g1 x200 f24000
g1 y200 f24000
g1 x0 f24000
g1 y0 f24000
g1 x200 f24000
g1 y200 f24000
g1 x0 f24000
g1 y0 f24000
g1 x200 f24000
g1 y200 f24000
```
Sometimes the last few commands will either be skipped, or more often, misinterpreted.

VanessaE, 2018-03-08:
That said, macros don't work:

```gcode
G91
G1 Z30 F1800
G90
G28 XY
G1 X100 Y100 F18000
M140 S0
M104 S0
M106 S255
M84
```

Assigned to a button named "Cool Down".  Click the button and:

```
Failed to handle button
Traceback (most recent call last):
  File "/home/vanessa/RepRap/Printrun-2.x/printrun/pronterface.py", line 2124, in process_button
    self.onecmd(command)
  File "/usr/lib/python3.5/cmd.py", line 217, in onecmd
    return func(arg)
  File "/home/vanessa/RepRap/Printrun-2.x/printrun/pronsole.py", line 437, in <lambda>
    setattr(self.__class__, "do_" + self.cur_macro_name, lambda self, largs, macro = macro: macro(self, *largs.split()))
TypeError: 'NoneType' object is not callable
```

Incidentally, the [+] button at the bottom of the window lacks any symbol.

kliment, 2018-03-25:
So are long macros now working correctly or not?

VanessaE, 2018-03-25:
I was wrong, long macros are still not being fully-sent.  Turns out that console/manual input also suffers.

kliment, 2018-03-25:
Are they being saved correctly in the config file?

VanessaE, 2018-03-25:
Looks like it.

KevWal, 2024-06-24:
I have the same issue that I detail here:  https://github.com/kliment/Printrun/issues/1265#issuecomment-2187467664