kliment/Printrun #764: Temperature display partially hidden until window resize

VanessaE, 2016-08-20:
Referring specifically to the numeric temperature/duty cycle displays located below the two "thermometers".  Printrun is at d36e6910f29040c86a71aa8727bbd28328bb4231 but this issue has existed since at least de535f11b38b254ce3a71cbe8a04413e582501dd.

On launch:
[image]

After window resize (doesn't matter if the window is made bigger or smaller):
[image]
(note the `B@:0`)

rockstorm101, 2017-12-14:
Hi, @VanessaE, does this keep happening with the latest release?

VanessaE, 2017-12-14:
It seems fine now.

closed 2017-12-14 by VanessaE