kliment/Printrun #768: Fedora 24: pronterface UI is badly mangled / Gtk warnings about negative sizes

klpeders, 2016-08-27:
Fedora 24: pronterface UI is badly mangled / Gtk warnings about negative sizesBoth `printrun` git and the version in F24 (2015.03.10-4.fc24) have a seriously mangled UI (to the point of being unusable):

[image]

 (first run after removing ~/.pronsolerc)

The program continuously spew the 4 gtk messages below about unexpected negative values:

```
(pronterface.py:1793): Gtk-CRITICAL **: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton
(pronterface.py:1793): Gtk-WARNING **: Negative content width -12 (allocation 22, extents 17x17) while allocating gadget (node button, owner GtkButton)
(pronterface.py:1793): Gtk-WARNING **: Negative content width -15 (allocation 19, extents 17x17) while allocating gadget (node button, owner GtkButton)
(pronterface.py:1793): Gtk-WARNING **: Negative content width -17 (allocation 1, extents 9x9) while allocating gadget (node entry, owner GtkEntry)
```

Line 1793 doesn't seem to have any particular relation to these messages.

kliment, 2016-08-27:
Oh wow, that's terrible. Do other wx applications look correct?

On 08/27/2016 07:21 AM, klpeders wrote:
> Fedora 24: pronterface UI is badly mangled / Gtk warnings about
> negative sizesBoth `printrun` git and the version in F24
> (2015.03.10-4.fc24) have a seriously mangled UI (to the point of
> being [...]

hroncok, 2016-08-27:
We already have some patches in Fedora to make it work, could you please
open Bugzilla for the package at bugzilla.redhat.com, I'll ping the author
of the patches. Thanks

klpeders, 2016-08-28:
I am sorry, searching redhat bugzilla : 
    https://bugzilla.redhat.com/buglist.cgi?quicksearch=printrun
and
    https://bugzilla.redhat.com/buglist.cgi?quicksearch=pronterface
doesn't give any relevant results. Do you have a link?

hroncok, 2016-08-28:
No, there's no link yet. Go to
https://bugzilla.redhat.com/enter_bug.cgi?product=Fedora&version=24&component=printrun

rockstorm101, 2017-12-14:
Hi guys, @klpeders, @hroncok, could we consider this matter terminated?

hroncok, 2017-12-15:
I'll package the latest version and run some basic tests. Hopefully some time during the upcoming holidays, but no promises.

rockstorm101, 2018-03-02:
Hi @hroncok, what the status of this issue?

hroncok, 2018-03-02:
I still see a lot of

    Gtk-WARNING **: Negative content width -...

But the UI looks mostly fine

[image]

kliment, 2018-03-28:
@hroncok can we consider this resolved?

hroncok, 2018-03-28:
We can close this one and solve the rest in https://github.com/kliment/Printrun/issues/913

closed 2018-03-28 by hroncok