kliment/Printrun #770: Documentation of binding execution of scripts to events is poor

Bushmills, 2016-09-01:
I'm trying to execute a script when printing has finished. It appears to be possible, as it's stated in the documentation, where "run_script" is described, that it can be, for example, used to send an email when print is ready.
What I couldn't find anywhere I looked is how to trap a particular event. I have a script, which I want to execute when printing has finished, but don't know how and where to specify this. It is nice that the doc hints that it's possibly, but it fails at giving a clue how it's done. Only thing I can currently think of is by periodically polling some printer status, and respond to the change of status, but I doubt that's how it's meant to be done.

SparkyCola, 2016-12-28:
This should be possible through the Settings > Options > External Commands. There you can enter commands that will be executed for slicing, on start, on final (that's probably what your looking for), at a simarrange and when an occurring error.

Bushmills, 2016-12-28:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA256

Thanks for reply. I was more concerned about pronsole, which doesn't
have the pronterface configuration dialog system you suggest to use, as
I'm using pronsole in a "batched" / scripted environment mostly. In that
context I look for a way to allow execution to proceed after printing an
object has completed. Being able to exit pronterface could already be
enough, as a shell script could then simply proceed (for example, with
turning off printer power again - printers are connected to "gembird"
like sispm controlled devices, perfectly scriptable through line
sispmctl, through which they are also selectively powered on prior to
scripted launch of pronsole). Being able to bind a pronsole script to
"object completed" event was thought to enable me exiting pronsole.
Kindly Yours,
L.Schmidt

>
> This should be possible through the Settings > Options > External
Commands. There you can enter commands that will be executed for
slicing, on start, on final (that's probably what your looking for), at
a simarrange and when an occurring error.
<https://github.com/kliment/Printrun/issues/770#issuecomment-269404899>,
or mute the thread
<https://github.com/

-----BEGIN PGP SIGNATURE-----
Version: GnuPG v2

iQIcBAEBCAAGBQJYY8fHAAoJEG4w1YOe5n+uXWgQAONs5MUQyvEj1Kd/o7oZd/bV
rD5g0vA5muLIZv1lwjYHdbaEfGIvagE8PnnKHNpZWzb4lPz6D/AcxkK2q5GD5HzA
uxSk7mD92wHNiXg/zstpd2/r3JsS43i8ZMYNEMe73cntUPd6oExqpl6OVi5l236f
kMSLMFlEvxVq3zp8BOICVaeA7gRGMXvBEdvNo2YzCbow0x2evK1grS0AWYOna7wE
grhRCjj1sIKTRzTrjXWnMpQsDEJUMYC2lviTrb/31CHn5ORC+Nrb3YigjEUFkaGn
/Ml3BQpUAdfTF1OvDxijnYgPKSbSX/hZ7z0kKl0uZXgHSa09UilGwdu7+dMuU3iM
gHTUUmE1WMAqJt/wKl2Z27mttx45Kf6iHfDHY7rFeh7WEfTvjmgOYwxerA/nUIng
obXGfQSS2F3MWnVP+0F6EaN0Yqc8xOViPwDCKhNh8DKflJmu+B+JrP/sl2eeiuVC
1sxWoWYdPs2AnumVhwydhBTAoOYBRm2XXbPYOi4cb/ufE8denQNZAdWJfErJ52vw
X84OxM8mzgkVuBzQjwpbvyheoJKapcLpHvEOOTPjolyato8ErheqTQgCt1Icy/f9
olkSKtDsIX/yHa7VaMQCZOnonRU+YfvZZQiwkcpG2pfwpZ4xeMhVTibXmpT3OX/y
Dx7foguqwcpE2Kz7OKmQ
=EZrC
-----END PGP SIGNATURE-----