kliment/Printrun #772: RTL and multi linuguality for pronterface

nimaam, 2016-09-16:
Dear Kliment
I really enjoy pronterface, and hope to have a good future for this nice product, we are focusing on localization of this software, and translate the messages and fit to our local needs, so I look into the code and find out that there is no possibilities to have the language selection and RTL for this software, do you have any Idea to do it in your mind map to be in github as next steps by you or it should handle seperately by us.
Thanks for your support.

nimaam, 2016-09-17:
No body here to help me on RTL support???????????

rockstorm101, 2016-09-18:
Hi @nimaam,

The efforts on in internationalizing Printrun started back in 2011 and a way to collaborate on this matter has been defined since then [[1]](https://github.com/kliment/Printrun/blob/master/README.i18n).

Regarding RTL support, please forgive my ignorance here, @aboobed did try to translate the plater functionality to Arabic [[2]](https://github.com/kliment/Printrun/blob/master/locale/ar/LC_MESSAGES/plater.po), could this work be of any use for you?

Cheers,
Rock

[1] https://github.com/kliment/Printrun/blob/master/README.i18n
[2] https://github.com/kliment/Printrun/blob/master/locale/ar/LC_MESSAGES/plater.po

nimaam, 2016-09-18:
Hi Kliment
I translated the printrun via po files but the RTL is this case and yesterday I do some edit in code to make it happen and I am going to finalize these changes but the matter is no structural way for this, I want to know do you want to make it in standard way
Nima

nimaam, 2016-09-26:
Dear Kliment
I develop the RTL codes but I have a problem with make binary in windows system could you please help me on it??
Please
With best regards
Nima