kliment/Printrun #773: Kossel Mini won't respond to Pronterface

grimes27, 2016-09-17:
I have my printer connected to pronterface and it says the printer is online but when I hit any of the home buttons on pronterface my printer does not respond.  I'm pretty new to this so I get lost pretty easy.  Any ideas?

Thanks in advance.

rockstorm101, 2016-09-18:
Hi @grimes27 ,

What messages are displayed in the log panel when you connect your printer? Could you please paste them? For instance, I get the following:

```
Connecting...
start
Printer is now online.
echo:Marlin1.0.2
echo: Last Updated: Mar 13 2016 19:20:39 | Author: Rock Storm
Compiled: Mar 13 2016
echo: Free Memory: 5220  PlannerBufferBytes: 1232
echo:Hardcoded Default Settings Loaded
echo:Steps per unit:
echo:  M92 X80.00 Y80.00 Z4000.00 E775.00
echo:Maximum feedrates (mm/s):
echo:  M203 X300.00 Y300.00 Z1.00 E45.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X5000 Y5000 Z10 E5000
echo:Acceleration: S=acceleration, T=retract acceleration
echo:  M204 S1000.00 T2000.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X20.00 Z0.40 E5.00
echo:Home offset (mm):
echo:  M206 X0.00 Y0.00 Z0.00
echo:PID settings:
echo:   M301 P38.96 I2.95 D128.46
```

grimes27, 2016-09-18:
Thank your for the quick response.  I am new to this and it's nice to see people are willing to help.

I did get the printer to respond, I'm pretty sure I had the wrong firmware loaded.  But I figured that out only to get stopped two steps into the woods on trying to configure the X-Axis, so I posted a new issue.  Any help would be appreciated.  Thanks you.

https://github.com/kliment/Printrun/issues/774

closed 2016-10-14 by grimes27