kliment/Printrun #776: Feature Planning: Progress view

SparkyCola, 2016-10-06:
Hi,
since most 3D-Printers have an LCD which is able to display at least 20 custom characters it would be useful it the display could display progress and the estimated remaining time. I'd like to code that and just want to be shure that my idea is worth realizing. There is one tiny problem though: MakerBots use M70 while Marlin, Repetier, Smoothie, RepRapFirmaware and Redeem use M117 (see [RepRapWiki](http://reprap.org/wiki/G-code#M117:_Display_Message)) which is used for getting the Zero Position by FiveD. How do I solve that issue? Would and option in the options be the right approach or is there a way in which Printrun detects the firmware?

kliment, 2016-10-07:
Hey, that's a nice idea. As far as I know there are quite a few issues
with makerbot firmware that we don't deal with, so I wouldn't worry
about makerbot compatibility.

On 10/07/2016 12:09 AM, Mattis M. wrote:
> Hi,
> since most 3D-Printers have an LCD which is able to display at least 20 custom characters it would be useful it the display could display progress and the estimated remaining time. I'd like to [...]

SparkyCola, 2016-10-07:
I appreciate your support and promise to work on that in the next week(s).

SparkyCola, 2016-10-07:
So for the frequency of the update would every minute be a good default value, right?

SparkyCola, 2016-10-12:
Got a first [working version](https://github.com/SparkyCola/Printrun). The only thing I'm wondering about is how I can send a `:` since this seems to be some kind of separator for commands.

paboman, 2016-10-13:
how does it work?

SparkyCola, 2016-10-13:
Works great

SparkyCola, 2016-12-28:
I've added a final version and I'll file a pull request any minute, the feature is now activate-able in the printer options, debugging console prints have been removed since they are not necessary.

rockstorm101, 2017-12-13:
I'm closing this issue. Fixed with 81298300f3d9a300a769dc503ba9691706f6fa68. Please feel free to reopen the issue if you think the matter is not solved.

closed 2017-12-13 by rockstorm101