kliment/Printrun #777: Feature Request - Filament Tracking

brownpurchase, 2016-10-08:
It would be really nice if I could tell printrun my spool size and let it track my filament length used/left.  Also, it could prevent me from trying to use a spool when the job is too long.  It would be nice to be able to save different spools as well.  So if I had 5 spools and I swapped out one color for another then it would save that I had 500ft left on blue and 200ft left on red.  I just have to select my active spool.  If that is too much then just the tracking would be best. 
Use Cases:
1. Add new spool - calculates the total length based upon weight and filament size or if no calc then #2.  puts it into the bin.  maybe a checkmark to make it current #4
2. Alter spool length - if the app doesn't track more than 1 spool I would need a way to alter the length after I add it.  Maybe I wasted more length for threading it in my system.
3. Store Spool - stores the spool in an offline status in the bin.
4. Use Spool - makes it the currently used spool in the bin.
5. Delete Spool - removes the spool from the bin.
6. Warn option - prevents or warns the user that their current print job is longer than spool has left.  
7. Buffer size - would add more to the spool calculation in #6 so that I don't get to the last yard to find out I am out.
8. logging - before each run it would log how much would be used and how much would be left.  same thing after the print was done.

rockstorm101, 2016-10-10:
I really like this idea, I've also felt the need for something like this, I'll see what I can do.

brownpurchase, 2016-10-13:
Thanks. I am excited to see how it turns out.

brownpurchase, 2016-10-29:
Another useful piece of information that could go with this is the temperature settings for the filament.  Just for reference.

SparkyCola, 2016-12-28:
This will have some offset (besides issues where the printer extrudes more or less than "planned"), this would be when you do anything on the printer screen (f.e. un-/loading filament on an Prusa i3 in order to change the spool) plus things like filament change via G-CODE without further options use firmware-defined lengths, anyway the effects should be considerable low. But it's still a nice idea. Just one last thing - How would you track the spool change?

dlarue, 2016-12-28:
Maybe the first thing would be to get the firmware(s) to support filament tracking/sensing and create standardized GCode for status, reset and reaction on error( ie Extruder motor commanded by filament tracking is not showing movement ).

SparkyCola, 2016-12-28:
Great idea, there is already one thing implemented (at least in the RepRap-Firmware): cold exctrusion prevented.

rockstorm101, 2017-06-21:
I've finally had the time to try to implement this. It'd be nice if you could take look at it [[1]].

[1]: https://github.com/kliment/Printrun/issues/827

brownpurchase, 2017-06-22:
I definitely will tonight. Thanks!

PR #827 (2017-06-28): Add a Spool Manager module
Dear all,

I've finally finished the implementation of what I've named a "Spool Manager". Although it is still in an alpha state, please try it out and let me know what you think.

The module is accessed through the GUI: Settings > Spool Manager.

I believe it does pretty much everything suggested initially by @brownpurchase (it complies with the 8 points of the "specification"). However it does not handle temperature settings for now.

I would love to hear any comments, suggestions, missing features, missing functionality,... Any feedback would be much appreciated.
 [... cut]
commit 5db1366c (2017-06-28): Add a Spool Manager module

(Closes: #777)
closed 2017-06-28 by kliment