kliment/Printrun #780: installation

Gotgot59, 2016-11-12:
Hello,
I had make the installation of printrun on my computer linux via the repository of  ppa:richi-paraeasy/ppa
and a mistake happen.

I don't know if it's here i have to make the request of progressing of the software installation of printrun.

it's just two little change in the python page : pronterface.py and pronsole.py 

the line : os.chdir(os.path.realpath(__file__)) have to be replace by : os.chdir(os.path.realpath(‘./’)) 

For a better usage, do you think is possible to make the change on the repository directly ? 

cf methode  :https://jetervaz.wordpress.com/2015/07/25/installing-pronterface-on-ubuntu-and-the-error-not-a-directory-usrshareprintrunpronterface-py/

kliment, 2016-11-12:
Unfortunately the problem is introduced by the installation package at
that ppa, and so fixing it upstream in our github repository will not
help. You will have to ask the ppa owner to fix it there.

On 11/12/2016 08:52 PM, Gautier wrote:
> Hello,
> I had make the installation of printrun on my computer linux via the repository of  ppa:richi-paraeasy/ppa
> and a mistake happen.
>
> I don't know if it's here i have to make the request of [...]

rockstorm101, 2016-11-13:
Hi @Gotgot59,

I deduct from your link that you are talking here about the installation of printrun in Ubuntu. Since Ubuntu 16.10 Yakkety Yak, Printrun is in the [official repositories](http://packages.ubuntu.com/yakkety/printrun), which means it can be installed by simply running:

`apt-get install printrun`

and there is no need to fetch it from a PPA. Please let me know if you encounter any trouble installing it this way.

Cheers,
Rock

Gotgot59, 2016-11-14:
Ok thanks for having taking the time to answer me ! 
Yes rockstorm101, i'm still in ubuntu 14.04 . ( I know i have to upgrade my os but ... don't want to take the time for restarting proprely an réinstall all the tools i have now. 
Is great we can now add print run in the official repository !

rockstorm101, 2017-12-13:
Problem seems to have been solved. I'm closing this issue.

closed 2017-12-13 by rockstorm101