kliment/Printrun #782: No "nagivation UI" on latest Fedora 24 update

bit4man, 2016-11-23:
The pronterface UI no longer has the navigation features in the top left corner after the latest Fedora 24 updates. I looked for newer updates than 2015 but it looks like that's the latest, so I wonder if Pronterface is even active anymore?
[image]

hroncok, 2016-11-23:
I have the same issue on Fedora 25, just upgraded form Fedora 23. Don't know what the issue is.

hroncok, 2016-11-23:
Actaully, there is a rectangle over the controls. This is upstream Printrun without our GTK3 patches.

[image]

kliment, 2016-11-23:
Any errors in console on startup? It smells like a path issue to me,
being unable to find the graphics. Are the buttons still there, just
invisible? What happens if you click on the area?

On 23.11.2016 15:40, Miro Hrončok wrote:
> I have the same issue on Fedora 25, just upgraded form Fedora 23. Don't know what the issue is.
>

hroncok, 2016-11-23:
With our GTK3 patch, no errors or warnings.

I cannot connect to a printer now to test clicking.

hroncok, 2016-11-23:
CC @swt2c

hroncok, 2016-11-23:
@bit4man What came in "the latest Fedora 24 updates"?

bit4man, 2016-11-23:
i have some Gtk warnings - but no more than usual:
$ pronterface

(pronterface:2511): Gtk-WARNING **: Negative content width -12 (allocation 22, extents 17x17) while allocating gadget (node button, owner GtkButton)

(pronterface:2511): Gtk-WARNING **: Negative content width -15 (allocation 19, extents 17x17) while allocating gadget (node button, owner GtkButton)

(pronterface:2511): Gtk-WARNING **: Negative content width -12 (allocation 22, extents 17x17) while allocating gadget (node button, owner GtkButton)

(pronterface:2511): Gtk-WARNING **: Negative content width -12 (allocation 22, extents 17x17) while allocating gadget (node button, owner GtkButton)

(pronterface:2511): Gtk-WARNING **: Negative content width -12 (allocation 22, extents 17x17) while allocating gadget (node button, owner GtkButton)

I can connect to a printer, and load gcode/stls etc. I just cannot navigate the printer movements manually. Clicking in the open area doesn't seem to work at all. Cursor doesn't change indicating you can click etc.

bit4man, 2016-11-23:
@hroncok:
Transaction: /13970_aabeecac
 System time: 2016-11-22T11:34:30.652395Z
 Succeeded: True
 Role: update-packages
 Duration: 161585 (seconds)
 Command line: 
 User ID: 0
 Username: root
 Real name: root
 Affected packages:
 - updating wxBase3-3.0.2-29.fc24.x86_64
 - installing kernel-core-4.8.8-200.fc24.x86_64
 - installing kernel-modules-4.8.8-200.fc24.x86_64
 - updating wxGTK3-3.0.2-29.fc24.x86_64
 - updating xorg-x11-server-common-1.18.4-5.fc24.x86_64
 - updating openssh-7.2p2-14.fc24.x86_64
 - updating libdrm-2.4.73-1.fc24.x86_64
 - installing kf5-knewstuff-5.27.0-1.fc24.x86_64
 - installing cln-1.3.4-4.fc24.x86_64
 - installing libqalculate-0.9.7-17.fc24.x86_64
 - updating xen-licenses-4.6.4-1.fc24.x86_64
 - updating selinux-policy-3.13.1-191.21.fc24.noarch
 - updating redhat-lsb-submod-multimedia-4.1-33.fc24.x86_64
 - updating redhat-lsb-submod-security-4.1-33.fc24.x86_64
 - updating redhat-lsb-core-4.1-33.fc24.x86_64
 - updating redhat-lsb-cxx-4.1-33.fc24.x86_64
 - updating redhat-lsb-desktop-4.1-33.fc24.x86_64
 - updating redhat-lsb-languages-4.1-33.fc24.x86_64
 - updating redhat-lsb-printing-4.1-33.fc24.x86_64
 - updating quota-nls-1:4.03-4.fc24.noarch
 - updating pciutils-libs-3.5.2-1.fc24.x86_64
 - updating magic-8.1.113-1.fc24.x86_64
 - updating libsndfile-1.0.27-1.fc24.x86_64
 - updating blender-fonts-1:2.78a-2.fc24.noarch
 - updating bind99-license-9.9.9-4.P4.fc24.noarch
 - installing libgit2-0.24.3-1.fc24.x86_64
 - installing kf5-ktexteditor-5.27.0-1.fc24.x86_64
 - installing kf5-kpty-5.27.0-4.fc24.x86_64
 - installing analitza-16.08.2-1.fc24.x86_64
 - installing luajit-2.0.4-4.fc24.x86_64
 - installing cantor-16.08.2-1.fc24.x86_64
 - installing cantor-libs-16.08.2-1.fc24.x86_64
 - updating LabPlot-2.3.0-6.fc24.x86_64
 - updating bind99-libs-9.9.9-4.P4.fc24.x86_64
 - updating blender-1:2.78a-2.fc24.x86_64
 - updating magic-doc-8.1.113-1.fc24.x86_64
 - updating pciutils-3.5.2-1.fc24.x86_64
 - updating quota-1:4.03-4.fc24.x86_64
 - updating redhat-lsb-4.1-33.fc24.x86_64
 - updating selinux-policy-targeted-3.13.1-191.21.fc24.noarch
 - updating xen-libs-4.6.4-1.fc24.x86_64
 - updating xorg-x11-server-Xorg-1.18.4-5.fc24.x86_64
 - updating xorg-x11-server-Xwayland-1.18.4-5.fc24.x86_64
 - updating openssh-server-7.2p2-14.fc24.x86_64
 - updating openssh-clients-7.2p2-14.fc24.x86_64
 - updating wxGTK3-media-3.0.2-29.fc24.x86_64
 - updating wxGTK3-gl-3.0.2-29.fc24.x86_64
 - installing kernel-4.8.8-200.fc24.x86_64
 - installing kernel-modules-extra-4.8.8-200.fc24.x86_64
 - updating xcircuit-3.9.58-1.fc24.x86_64
 - updating python3-dateutil-1:2.6.0-1.fc24.noarch
 - updating python2-dateutil-1:2.6.0-1.fc24.noarch
 - updating python2-cryptography-1.5.3-3.fc24.x86_64
 - updating python-nss-1.0.0-2.fc24.x86_64
 - updating python-nose-1.3.7-9.fc24.noarch
 - updating perl-podlators-4.09-1.fc24.noarch
 - updating pcre2-10.21-9.fc24.x86_64
 - updating kernel-headers-4.8.8-200.fc24.x86_64
 - updating irsim-9.7.95-1.fc24.x86_64
 - updating flatpak-libs-0.6.13-2.fc24.x86_64
 - updating firefox-50.0-1.fc24.x86_64
 - updating dnsmasq-2.76-2.fc24.x86_64
 - updating breeze-icon-theme-5.27.0-2.fc24.noarch
 - updating augeas-libs-1.7.0-1.fc24.x86_64
 - installing kernel-devel-4.8.8-200.fc24.x86_64

That's the updates package kit did when I shutdown the night before last. This morning this is the result I saw (didn't have the system on yesterday).

swt2c, 2016-11-23:
Someone reported a regression in Poedit also after the latest wxGTK3 update.  I've got a fix for that one in hand.  I'll check tonight whether it fixes Printrun too and get it out ASAP.

swt2c, 2016-11-24:
Confirmed, the same patch that fixes Poedit fixes Printrun too.

swt2c, 2016-11-24:
https://bodhi.fedoraproject.org/updates/FEDORA-2016-74bef396a8

hroncok, 2016-11-24:
Fix confirmed.

bit4man, 2016-11-24:
Somehow bodhi/updates-testing isn't pulling it down for me to verify?

swt2c, 2016-11-24:
It hasn't been pushed to updates-testing yet.  AFAIK, updates pushes in Fedora still require human intervention to sign the packages, so I'm not sure if there is anyone around to push updates today.

If you want to try it immediately, you can download the packages directly from Koji:
http://koji.fedoraproject.org/koji/buildinfo?buildID=820092

bit4man, 2016-11-24:
Thanks @swt2c. That did it :) I did notice bodhi said the packages were signed but I'm impatient :) Need to calibrate the extruder steps and it's hard to do straight on the firmware :)

rockstorm101, 2017-12-13:
Hi guys, @bit4man, @hroncok and @swt2c, could we consider this matter terminated and close this issue?

swt2c, 2017-12-13:
Yes, I believe so.  This was a wxGTK issue anyway, not a printrun one.

closed 2017-12-14 by rockstorm101