kliment/Printrun #783: Fedora 25 pronterface.rpm does not include "en" pronterface.mo?

bit4man, 2016-11-30:
The latest pronterface.rpm for Fedora 25 (2015.03.10) seems to be missing the english language files:
[peter@boss ~]$ rpm -ql pronterface
/usr/bin/pronterface
/usr/share/appdata/pronterface.appdata.xml
/usr/share/applications/pronterface.desktop
/usr/share/doc/pronterface
/usr/share/doc/pronterface/COPYING
/usr/share/doc/pronterface/README.cleanup
/usr/share/doc/pronterface/README.i18n
/usr/share/doc/pronterface/README.md
/usr/share/locale/de/LC_MESSAGES/pronterface.mo
/usr/share/locale/fr/LC_MESSAGES/pronterface.mo
/usr/share/locale/it/LC_MESSAGES/pronterface.mo
/usr/share/locale/nl/LC_MESSAGES/pronterface.mo
/usr/share/pixmaps/pronterface.png
/usr/share/pronterface
/usr/share/pronterface/images
/usr/share/pronterface/images/arrow_down.png
/usr/share/pronterface/images/arrow_keys.png
/usr/share/pronterface/images/arrow_up.png
/usr/share/pronterface/images/control_mini.png
/usr/share/pronterface/images/control_mini.svg
/usr/share/pronterface/images/control_xy.png
/usr/share/pronterface/images/control_xy.svg
/usr/share/pronterface/images/control_z.png
/usr/share/pronterface/images/control_z.svg
/usr/share/pronterface/images/control_z_mini.png
/usr/share/pronterface/images/control_z_mini.svg
/usr/share/pronterface/images/edit.png
/usr/share/pronterface/images/fit.png
/usr/share/pronterface/images/inject.png
/usr/share/pronterface/images/reset.png
/usr/share/pronterface/images/zoom_in.png
/usr/share/pronterface/images/zoom_out.png
/usr/share/pronterface/locale

As a result, when I run "pronterface" I'm coredumped when trying to launch pronterface:
stat("/usr/share/pronterface/locale", {st_mode=S_IFDIR|0755, st_size=20480, ...}) = 0
stat("/usr/share/pronterface/locale/en_US.UTF-8/LC_MESSAGES/pronterface.mo", 0x7fffd1acd5d0) = -1 ENOENT (No such file or directory)
stat("/usr/share/pronterface/locale/en_US/LC_MESSAGES/pronterface.mo", 0x7fffd1acd5d0) = -1 ENOENT (No such file or directory)
stat("/usr/share/pronterface/locale/en.UTF-8/LC_MESSAGES/pronterface.mo", 0x7fffd1acd5d0) = -1 ENOENT (No such file or directory)
stat("/usr/share/pronterface/locale/en/LC_MESSAGES/pronterface.mo", 0x7fffd1acd5d0) = -1 ENOENT (No such file or directory)
close(14)                               = 0
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0x1} ---
+++ killed by SIGSEGV (core dumped) +++
Segmentation fault (core dumped)

bit4man, 2016-11-30:
Sorry - here's the actual package details:
$ rpm -qi pronterface
Name        : pronterface
Version     : 2015.03.10
Release     : 5.fc25
Architecture: noarch
Install Date: Wed 30 Nov 2016 11:42:48 AM EST
Group       : Applications/Engineering
Size        : 365661
License     : GPLv3+ and CC0
Signature   : RSA/SHA256, Wed 20 Jul 2016 03:20:36 AM EDT, Key ID 4089d8f2fdb19c98
Source RPM  : printrun-2015.03.10-5.fc25.src.rpm
Build Date  : Tue 19 Jul 2016 05:49:00 AM EDT
Build Host  : arm04-builder20.arm.fedoraproject.org
Relocations : (not relocatable)
Packager    : Fedora Project
Vendor      : Fedora Project
URL         : https://github.com/kliment/Printrun
Summary     : GUI interface for RepRap
Description :
Pronterface is a featured G-code sender with graphical user interface.
It controls the ReRap printer. It is a part of Printrun.

hroncok, 2016-11-30:
I'm not sure if English has language files. The package works for me. I have no idea where's the actuall problem.

bit4man, 2016-11-30:
Any ideas on how I can turn up debug to see what's causing the issue? Infortunately abrt isn't allowing me to report the issue since pronterface isn't found on the retrace server.

kliment, 2018-03-24:
Is this still a problem with latest version?