kliment/Printrun #785: Wayland (Fedora 25) causes Segfault?

bit4man, 2016-12-06:
I'm getting segfaults after upgrading to F25 when I launch pronterface. I do not get Pronterface loaded at all and abrt refuses to post a BZ since no frame or threads are found in the dump. From retrace it seems the issue is caused by "libwayland-client.so.0".  I need advice on how to provide the correct dump/data on this.

bit4man, 2016-12-06:
Dump created by abrt
[pronterface.zip](https://github.com/kliment/Printrun/files/634400/pronterface.zip)

hroncok, 2016-12-06:
If you run Pronterface with:

    GDK_BACKEND=x11 pronterface

Is the problem gone? We have had some crashes with wxPython apps on Wayland, so if this fixes the issue, I will make it the default in Fedora.

bit4man, 2016-12-06:
I don't have access to the actual workstation today. However, I did a ssh -X (X11 forwarding) to the box and that works. Which is why I concluded it was Wayland. Sorry for not stating that initially.

bit4man, 2016-12-07:
@hroncok confirmed. Setting GDK_BACKEND=x11 fixes the issue (or bypasses it to be more specific). Thanks for the tip!

gbr1918, 2016-12-13:
@hroncok  Great!  That's fixed my problem - thank you very much indeed, Hroncok.

rockstorm101, 2017-12-13:
I'm closing this issue. Fixed with 6a7c9d7f5f68fa2513836590111f07d19ab01536. Please feel free to reopen the issue if you think the matter is not solved.

closed 2017-12-13 by rockstorm101