kliment/Printrun #786: Printrun Compiled windows

kenoro, 2016-12-07:
Hi,

I tried to compile and use the master branch on windows but I get errors on the start.

Could anybody make a useable program for windows from the master brach?

Anyway the errors are:

WARNING:root:Memory-efficient GCoder implementation unavailable: No module named
 gcoder_line
WARNING:root:psutil unavailable, could not import power utils:No module named ps
util
[ERROR] 3D view mode requested, but we failed to initialize it.
Falling back to 2D view, and here is the backtrace:
Traceback (most recent call last):
  File "C:\Program Files (x86)\Printrun-master\printrun\gui\viz.py", line 70, in
 __init__
    import printrun.gcview
  File "C:\Program Files (x86)\Printrun-master\printrun\gcview.py", line 22, in
<module>
    from .gl.panel import wxGLPanel
  File "C:\Program Files (x86)\Printrun-master\printrun\gl\panel.py", line 21, i
n <module>
    import numpy
ImportError: No module named numpy

Traceback (most recent call last):
  File "C:\Program Files (x86)\Printrun-master\pronterface.py", line 63, in <mod
ule>
    app = PronterApp(False)
  File "C:\Program Files (x86)\Printrun-master\printrun\pronterface.py", line 22
22, in __init__
    self.mainwindow = PronterWindow(self)
  File "C:\Program Files (x86)\Printrun-master\printrun\pronterface.py", line 21
8, in __init__
    self.reload_ui()
  File "C:\Program Files (x86)\Printrun-master\printrun\pronterface.py", line 28
0, in reload_ui
    self.settings.controlsmode == "Mini")
  File "C:\Program Files (x86)\Printrun-master\printrun\gui\__init__.py", line 2
56, in createGui
    viz_pane = VizPane(self, vizpanel)
  File "C:\Program Files (x86)\Printrun-master\printrun\gui\viz.py", line 79, in
 __init__
    from printrun import gviz
  File "C:\Program Files (x86)\Printrun-master\printrun\gviz.py", line 18, in <m
odule>
    import numpy
ImportError: No module named numpy

Thank you for helping me

rockstorm101, 2016-12-07:
You've probably already noticed this but just in case it helps, looking at 

> ImportError: No module named numpy

it seems it cannot find [numpy](http://www.numpy.org/). Either it is not installed at all or it is installed on a place where printrun is unable to locate it. If you were in the second case, [here](http://www.scipy.org/scipylib/building/windows.html#id13) I found a way to know where numpy should be installed to be found properly.

Hope this helps.

rockstorm101, 2017-12-13:
I'm closing this issue since a new release has been published.

closed 2017-12-13 by rockstorm101