kliment/Printrun #787: Can't Print; maybe a GCode Issue

Codezilla02, 2016-12-08:
Hello, I was inquiring about an error I have all besuden came across. This is console I receive. Please I have a 3D designing class and I need to get it printed. I have tried re downloading as well. 

Slicing C:\Users\Cody\Downloads\phone_case (3).stl
Running Slic3r/slic3r.exe C:\Users\Cody\Downloads\phone_case (3).stl --load config.ini --output 
C:\Users\Cody\Downloads\phone_case (3)_export.gcode
Cannot find specified configuration file (config.ini).

Loading file: C:\Users\Cody\Downloads\phone_case (3)_export.gcode
Exception in thread Thread-5:
Traceback (most recent call last):
  File "threading.pyc", line 808, in __bootstrap_inner
  File "threading.pyc", line 761, in run
  File "printrun\pronterface.pyc", line 1368, in load_gcode_async_thread
  File "printrun\pronsole.pyc", line 886, in load_gcode
IOError: [Errno 2] No such file or directory: u'C:\\Users\\Cody\\Downloads\\phone_case (3)_export.gcode'

kliment, 2016-12-08:
Looks like your slic3r config file is missing. If you're in a hurry,
open the file in slic3r directly, export a gcode file and then open that
in pronterface. To fix the error permanently, adjust your settings in
slic3r then export then to a file called config.ini and save it in the
printrun directory

On 08.12.2016 03:54, Codezilla02 wrote:
> Hello, I was inquiring about an error I have all besuden came across. This is console I receive. Please I have a 3D designing class and I need to get it printed. I have tried re downloading as [...]

rockstorm101, 2017-12-13:
Hi, @Codezilla02, did kliment's answer solve your problem? If so, would you consider closing the issue?

closed 2018-03-02 by rockstorm101