kliment/Printrun #789: Pronterface crashes on Fedora 25

gbr1918, 2016-12-09:
Hi,
I've been using Pronterface on Fedora 24 and it's been fine.  Since upgrading to Fedora 25, Pronterface crashes when I try to open it.   Here's the relevant journalctl -r  output (latest at the top):

Dec 09 22:21:47 humbert.Accura abrt-server[8468]: Deleting problem directory ccpp-2016-12-09-22:21:46-8441 (dup of ccpp-2016-12-08-09:55
Dec 09 22:21:46 humbert.Accura kernel: perf: interrupt took too long (3920 > 3918), lowering kernel.perf_event_max_sample_rate to 51000
Dec 09 22:21:46 humbert.Accura abrt-hook-ccpp[8467]: Process 8441 (python2.7) of user 1000 killed by SIGSEGV - dumping core
Dec 09 22:21:46 humbert.Accura kernel: pronterface[8441]: segfault at 1 ip 00007fc12589d52e sp 00007ffeac0c5b10 error 4 in libX11.so.6.3
Dec 09 22:21:46 humbert.Accura kernel: show_signal_msg: 198 callbacks suppressed
Dec 09 22:21:46 humbert.Accura audit[8441]: ANOM_ABEND auid=1000 uid=1000 gid=1000 ses=2 pid=8441 comm="pronterface" exe="/usr/bin/pytho
Dec 09 22:21:45 humbert.Accura pronterface.desktop[8441]: WARNING:root:Could not setup DBus for sleep inhibition: No module named dbus

Any help would be very welcome!

Dave Cooper

hroncok, 2016-12-09:
See https://github.com/kliment/Printrun/issues/785#issuecomment-265277861

bit4man, 2016-12-18:
Using the 3D viewer will eventually crash nouveau:
Dec 18 13:38:54 boss.peterlarsen.org kernel: nouveau 0000:05:00.0: gr: TRAP ch 7 [00bef11000 Xwayland[1937]]
Dec 18 13:38:54 boss.peterlarsen.org kernel: nouveau 0000:05:00.0: gr: GPC0/TPC1/MP trap: global 00000004 [MULTIPLE_WARP_ERRORS] warp 1000d [OOR_REG]
Dec 18 13:38:54 boss.peterlarsen.org kernel: nouveau 0000:05:00.0: gr: GPC2/PROP trap: 00000080 [ZETA_STORAGE_TYPE_MISMATCH] x = 32, y = 104, format = 0, storage type = fe
Dec 18 13:38:54 boss.peterlarsen.org kernel: nouveau 0000:05:00.0: gr: GPC3/TPC0/MP trap: global 00000004 [MULTIPLE_WARP_ERRORS] warp 000d [OOR_REG]
Dec 18 13:38:54 boss.peterlarsen.org kernel: nouveau 0000:05:00.0: gr: GPC4/TPC0/MP trap: global 00000004 [MULTIPLE_WARP_ERRORS] warp 000d [OOR_REG]
Dec 18 13:38:54 boss.peterlarsen.org kernel: nouveau 0000:05:00.0: gr: GPC4/TPC2/MP trap: global 00000004 [MULTIPLE_WARP_ERRORS] warp 2000d [OOR_REG]
Dec 18 13:38:59 boss.peterlarsen.org kernel: nouveau 0000:05:00.0: fifo: SCHED_ERROR 0a [CTXSW_TIMEOUT]
Dec 18 13:38:59 boss.peterlarsen.org kernel: nouveau 0000:05:00.0: fifo: gr engine fault on channel 10, recovering...
Dec 18 13:39:03 boss.peterlarsen.org kernel: nouveau 0000:05:00.0: fifo: SCHED_ERROR 0a [CTXSW_TIMEOUT]
Dec 18 13:39:03 boss.peterlarsen.org kernel: nouveau 0000:05:00.0: fifo: gr engine fault on channel 6, recovering...
Dec 18 13:39:45 boss.peterlarsen.org kernel: nouveau 0000:05:00.0: Xwayland[1937]: nv50cal_space: -16

Using the 3D viewer to get feedback on where the print process is. Wayland will crash (stop responding) and the last line in the above dump is repeated over and over again. Force reboot is the only way I've found to recover.

hroncok, 2016-12-18:
Is this related?

bit4man, 2016-12-18:
@hroncok better people than I can probably answer that. To me, it's the same problem: After Fedora 25, Pronterface is having problems running. I cannot say if it's the same issue or not - but my presumption is that the work-around of using X11 isn't 100% working.  I don't mind creating another issue if that helps.

hroncok, 2016-12-18:
Actually, reporting it trough ABRT (Automatic Bug Reporting Tool) to Bugzilla would give me the best information.

rockstorm101, 2017-12-13:
I'm closing this issue. Fixed with 6a7c9d7f5f68fa2513836590111f07d19ab01536. Please feel free to reopen the issue if you think the matter is not solved.

closed 2017-12-13 by rockstorm101