kliment/Printrun #791: Why show EX1 Target and EX1 Temp in graph when a single extruder in use?

PJMoore, 2016-12-10:
Why is the temperature graph still showing EX1 data when a single extruder is designated in Settings/Options/Extruders count?

garybarnes19660630, 2017-06-29:
I'm having the same problem.

dwillmore, 2018-09-30:
I am experiencing this as well.  Firmware is Marlin 1.1.9  The response to M105 is:
ok T:22.50 /0.00 B:22.42 /0.00 @:0 B@:0
Ex1 shows a temp of just over 23 and a target of 0.  The values (actual and target) for goth Ex0 and Bed reflect the values in the M105 response.

Version of Printrun is 2014.08.01 as is current for Fedora 27.  I'd be glad to try newer versions if it is believed that will make a difference. 

Thank you for your work on this very useful program.

dwillmore, 2018-10-01:
I've looked into this a bit.  It looks like Printrun/graph.py is hardwired to handle a bed and two extruders.  There is a part which imples that zero values should cause ex1 not to display.  But for some reason that's not triggering because a false temp is being reported for ex1.

The problem seems to be that something is incorrectly populating the ex1 temperature data.  If I heat or cool the bed and ex0 thermistors, the graph responds apropriately.  Nothing seems to effect ex1--it stays constant the whole time.  The same goes for the set point of ex1.

I'm not a python programmer and I'm unfamiliar with the code base, but I'll keep poking around.

commit 28f753b1 (2018-11-16): Add setting for graph background color
Hide second extruder from graph if not present
Fixes #791
closed 2018-11-16 by kliment