kliment/Printrun #796: Printrun lists too generic MIME types

rockstorm101, 2017-01-12:
As originally reported by Christian Mock [1]: "text/plain" and "application/octet-stream" are listed as MIME types for pronterface (in addition to "application/sla", which is fine).
 
This means that depending on other packages installed etc, applications such as firefox offer pronterface as the first option to open text files or generic binary files.

[1] https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=850698#5

hroncok, 2017-01-12:
I see three options here:

## Keep it as it is

 * **Pros:** Pronterface is in the list of apps that can open STLs, GCODEs
 * **Cons:** Pronterface is in the list of apps that can open ordinary text files and generic binary files

## Remove those

 * **Pros:** Pronterface is not in the list of apps that can open ordinary text files and generic binary files
 * **Cons:** Pronterface is not in the list of apps that can open STLs, GCODEs

## Introduce a real MIME type for STLs and GCODEs in [shared-mime-info](https://www.freedesktop.org/wiki/Specifications/shared-mime-info-spec/), use those instead

* **Pros:** Pronterface is in the list of apps that can open STLs, GCODEs, but not in the list of apps that can open ordinary text files and generic binary files. Other apps suffering form this problem can benefit from this as well.
* **Cons:** More work than any of the previous two

My odl (never finished) first attempt: https://github.com/hroncok/shared-mime-info-3dprint/commit/207d3b5a58e83e96f3fce3ea373129d572ba9d2a

hroncok, 2017-01-18:
Those now mime types had been merged to *shared-mime-info*:

 * `model/x.stl-binary`
 * `model/x.stl-ascii`
 * `text/x.gcode`

PR #854 (2017-10-08): Remove pronterface generic mimetypes
Remove the generic mimetypes listed in 'pronterface.desktop' and replace them with the ones actually
associated to STL models and G-Code files. (Fixes: #796)
commit 268543df (2017-10-08): Remove pronterface generic mimetypes

Remove the generic mimetypes and replace them with the ones actually
associated to STL models and G-Code files. (Fixes: #796)
closed 2017-10-08 by kliment