kliment/Printrun #798: Raspberry Pi and pronsole

voidloop, 2017-01-24:
After command "connect", I obtain this error: 

```
offline> connect
No port specified - connecting to /dev/ttyACM0 at 115200bps
offline> [ERROR] Traceback (most recent call last):
  File "/home/pi/workspace/3d/Printrun/printrun/printcore.py", line 243, in _readline
    try: self.recvcb(line)
  File "/home/pi/workspace/3d/Printrun/printrun/pronsole.py", line 1272, in recvcb
    if self.silent is False: self.log("\r" + l.ljust(15))
  File "/home/pi/workspace/3d/Printrun/printrun/pronsole.py", line 253, in log
    msg = u"".join(unicode(i) for i in msg)
  File "/home/pi/workspace/3d/Printrun/printrun/pronsole.py", line 253, in <genexpr>
    msg = u"".join(unicode(i) for i in msg)
  File "/usr/lib/python2.7/encodings/utf_8.py", line 16, in decode
    return codecs.utf_8_decode(input, errors, True)
UnicodeDecodeError: 'utf8' codec can't decode byte 0xf0 in position 19: invalid continuation byte

Printer is now online
PC> exit
```

but after this error pronsole works correctly. 
The same version of pronsole under Fedora 24 works without any error. The version is the master branch.

JimBarrows, 2017-02-07:
I'm getting almost the same error:
`[ERROR] Traceback (most recent call last):
  File "/home/jim/Desktop/Printrun/printrun/printcore.py", line 243, in _readline
    try: self.recvcb(line)
  File "/home/jim/Desktop/Printrun/printrun/pronsole.py", line 1272, in recvcb
    if self.silent is False: self.log("\r" + l.ljust(15))
  File "/home/jim/Desktop/Printrun/printrun/pronsole.py", line 253, in log
    msg = u"".join(unicode(i) for i in msg)
  File "/home/jim/Desktop/Printrun/printrun/pronsole.py", line 253, in <genexpr>
    msg = u"".join(unicode(i) for i in msg)
  File "/usr/lib/python2.7/encodings/utf_8.py", line 16, in decode
    return codecs.utf_8_decode(input, errors, True)
UnicodeDecodeError: 'utf8' codec can't decode byte 0x93 in position 2: invalid start byte`

OS: Ubuntu 16.04
Printer: BCN3d Sigma.

rockstorm101, 2017-12-13:
Hi, @voidloop and @JimBarrows:

 * Does this problem happen also with the latest release?

 * Did you try the fix I proposed?

Thanks

rockstorm101, 2018-03-02:
I'm closing this issue due to a lack of feedback. Please feel free to reopen if needed.

closed 2018-03-02 by rockstorm101

volconst, 2019-04-14:
I had similar problem. It turned out that the baudrate was not 250000. I ran pronterface, selected 250000, connected, exited pronterface and then pronsole worked fine.