kliment/Printrun #799: GCoder implementation unavailable

paintbot, 2017-01-28:
im running mint linux 18.1 Cinnamon 32-bit and having problems with the famous error:

WARNING:root:Memory-efficient GCoder implementation unavailable: No module named gcoder_line
running build_ext

when running pronterface.py 

I have cython installed but not sure its working probably as im a linux noob.

any help appreciated! 

from 

python setup.py build_ext --inplace

i get the following error:

paintbot@paintbot ~/Printrun-master $ python setup.py build_ext --inplace
Compiling printrun/gcoder_line.pyx because it changed.
[1/1] Cythonizing printrun/gcoder_line.pyx
WARNING:root:Memory-efficient GCoder implementation unavailable: No module named gcoder_line
running build_ext
building 'printrun.gcoder_line' extension
creating build
creating build/temp.linux-i686-2.7
creating build/temp.linux-i686-2.7/printrun
i686-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I/usr/include/python2.7 -c printrun/gcoder_line.c -o build/temp.linux-i686-2.7/printrun/gcoder_line.o
printrun/gcoder_line.c:12:20: fatal error: Python.h: No such file or directory
compilation terminated.
error: command 'i686-linux-gnu-gcc' failed with exit status 1

hroncok, 2017-01-29:
> `printrun/gcoder_line.c:12:20: fatal error: Python.h: No such file or directory`

You don't have Python header files. Get them from your distribution package manager. On Fedora, for example , that would be `sudo dnf install python2-devel`, or better yet just do `sudo dnf install pronterface` if you need to use it and don't need to run the git master branch or your custom changes. On other distributions the commands might be different.

rockstorm101, 2017-01-29:
On Linux Mint that would be `sudo apt-get install libpython-dev`.

This is somehow a duplicate of #763.

paintbot, 2017-01-29:
thanks guys! 

I have installed some more libraries including libpython-dev

now i get a different message but it still seems not proper:

python setup.py build_ext --inplace
WARNING:root:Memory-efficient GCoder implementation unavailable: No module named gcoder_line
running build_ext
building 'printrun.gcoder_line' extension
i686-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I/usr/include/python2.7 -c printrun/gcoder_line.c -o build/temp.linux-i686-2.7/printrun/gcoder_line.o
i686-linux-gnu-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,relro -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -Wdate-time -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wl,-Bsymbolic-functions -Wl,-z,relro -Wdate-time -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security build/temp.linux-i686-2.7/printrun/gcoder_line.o -o /home/paintbot/Printrun-master/printrun/gcoder_line.so

paintbot, 2017-02-02:
works now! sorry I'm a noob :)

closed 2017-02-02 by paintbot