kliment/Printrun #801: Where are macros in Pronterface software?

soosalameh, 2017-02-02:
Hi, so I want to edit the software such that instead of having users enter macros/Gcode commands for custom buttons themselves, that I can have specific custom buttons already included in the software for when they download it.

However, I can't find where the macros are in the software. Could someone help me with this?

soosalameh, 2017-02-02:
Update: I found that the macros are saved in the printrunconf.ini file under C:\Users\(Username) but want to edit this file before download. Anyone know where to find it in the "Printrun-Win-Slic3r-03Feb2015" folder?

rockstorm101, 2017-12-13:
Hi, such file is not distributed, it will be automatically generated, if it doesn't exist, the first time you run pronsole/pronterface.

rockstorm101, 2018-02-24:
I'm closing this issue. The question was answered and the OP did not complain. Please feel free to reopen if needed.

closed 2018-02-24 by rockstorm101