kliment/Printrun #802: Pronterface errors

Mypantera, 2017-02-09:
Hi all,

Just installed pronterface using the latest versions and keep getting these errors.  I deleted and reinstalled it several times with no changes.  I know the printer itself is fine because it was working before at the guys house who rebuild it.  I was using an Azteeg X1 but when that died the original builder installed a smoothie board and was using pronterface.  It was printing fine when it left his place.

The errors I keep getting are:
Printer is now online.
[ERROR] Traceback (most recent call last):
  File "printrun\printcore.pyc", line 241, in _readline
  File "printrun\pronterface.pyc", line 1713, in recvcb
  File "printrun\pronterface.pyc", line 1669, in update_pos
ValueError: could not convert string to float: 

Board: smoothie
Pronterface version: printrun 2014.08.01

another note: when I look at the files in the printrun folder, I donot see any that end in .pyc, mine all end in .py (example printcore.py, instead of printcore.pyc)
I am not ever sure where to begin looking... any help is greatly appreciated.

Mypantera, 2017-02-11:
resolved
had to reinstall the firmware and it started working

rockstorm101, 2017-02-11:
Nice to hear you solved this weird errors, would you consider closing this issue if it is in fact solved?

closed 2017-02-11 by Mypantera