kliment/Printrun #805: Tabbed with platter interface mode crashes pronterface

foockinho, 2017-02-21:
when selecting tabbed with platter mode, pronterface crashes and yields this error

Traceback (most recent call last):
  File "/home/foockinho/software/Printrun-master/printrun/pronterface.py", line 348, in kill
    wx.CallAfter(self.gwindow.Destroy)
  File "/usr/lib/python2.7/dist-packages/wx-3.0-gtk2/wx/_core.py", line 16710, in __getattr__
    raise PyDeadObjectError(self.attrStr % self._name)
wx._core.PyDeadObjectError: The C++ part of the GvizWindow object has been deleted, attribute access no longer allowed.

if i try to restart pronterface it desnt start and yields the following error

Traceback (most recent call last):
  File "pronterface.py", line 63, in <module>
    app = PronterApp(False)
  File "/home/foockinho/software/Printrun-master/printrun/pronterface.py", line 2222, in __init__
    self.mainwindow = PronterWindow(self)
  File "/home/foockinho/software/Printrun-master/printrun/pronterface.py", line 218, in __init__
    self.reload_ui()
  File "/home/foockinho/software/Printrun-master/printrun/pronterface.py", line 277, in reload_ui
    self.createTabbedGui()
  File "/home/foockinho/software/Printrun-master/printrun/gui/__init__.py", line 155, in createTabbedGui
    add_extra_controls(extracontrols, self, page1panel2, controls_sizer.extra_buttons)
  File "/home/foockinho/software/Printrun-master/printrun/gui/controls.py", line 299, in add_extra_controls
    add("tempgraph", root.graph, flag = wx.EXPAND | wx.ALL, border = 5)
  File "/home/foockinho/software/Printrun-master/printrun/gui/controls.py", line 126, in add
    container.Add(widget, *args, **kwargs)
  File "/usr/lib/python2.7/dist-packages/wx-3.0-gtk2/wx/_core.py", line 15931, in Add
    return _core_.GridBagSizer_Add(*args, **kwargs)
wx._core.PyAssertionError: C++ assertion "!CheckForIntersection(item)" failed at ../src/common/gbsizer.cpp(223) in Add(): An item is already at that position

A workaround ive found is to manually edit the file .pronsolerc stored at home directory changing the last line:

set uimode Tabbed with platter

by

set uimode Standard

apelly, 2017-06-22:
Confirming this same issue:

11:43 $ Traceback (most recent call last):
  File "./pronterface.py", line 66, in <module>
    app = PronterApp(False)
  File "/home//src/Printrun/printrun/pronterface.py", line 2251, in __init__
    self.mainwindow = PronterWindow(self)
  File "/home//src/Printrun/printrun/pronterface.py", line 219, in __init__
    self.reload_ui()
  File "/home//src/Printrun/printrun/pronterface.py", line 278, in reload_ui
    self.createTabbedGui()
  File "/home//src/Printrun/printrun/gui/__init__.py", line 155, in createTabbedGui
    add_extra_controls(extracontrols, self, page1panel2, controls_sizer.extra_buttons)
  File "/home//src/Printrun/printrun/gui/controls.py", line 299, in add_extra_controls
    add("tempgraph", root.graph, flag = wx.EXPAND | wx.ALL, border = 5)
  File "/home//src/Printrun/printrun/gui/controls.py", line 126, in add
    container.Add(widget, *args, **kwargs)
  File "/usr/lib/python2.7/dist-packages/wx-3.0-gtk2/wx/_core.py", line 15931, in Add
    return _core_.GridBagSizer_Add(*args, **kwargs)
wx._core.PyAssertionError: C++ assertion "!CheckForIntersection(item)" failed at ../src/common/gbsizer.cpp(223) in Add(): An item is already at that position

rockstorm101, 2017-12-14:
Confirmed.
 * OS:      Windows 10 Pro
 * Version:          1.6.0

frhun, 2017-12-22:
Confirmed on Linux too

crystalline, 2017-12-31:
Confirmed on debian sid with packaged version
`printrun/unstable,testing,now 1.6.0-1 amd64 [installed]`
Restored to functional state by `sudo apt purge printrun; sudo apt install printrun`

timlyo, 2018-01-05:
I'm also getting this issue. Reinstalling isn't working for me, are there any config files I can reset?

Running on Arch with Version 1.6

rockstorm101, 2018-01-07:
Hi all, as @timlyo said, deleting the file works. A less drastic approach to restoring the standard interface is simply editing `~/.pronsolerc` from:
```
set uimode Tabbed
```
to:
```
set uimode Standard
```

kliment, 2018-03-19:
I'm going to disable tabbed mode entirely for the moment as it currently is fairly broken

kliment, 2018-03-21:
I'd leave this bug out of the 2.x milestone because tabbed mode is disabled and I think we can leave it that way for release.

rockstorm101, 2018-03-21:
Agree.

kliment, 2018-03-24:
This has now been fixed in 2.x - however the tabbed mode looks terrible and doesn't size the window correctly so I've enabled it but it's hidden in the GUI. Putting set uimode Tabbed in the config file will still enable tabbed mode for testing purposes for further development. I'm closing this issue for now as the original issue is now resolved in 2.x. The visual issues with tabbed mode in 2.x will get their own issue.

closed 2018-03-24 by kliment