kliment/Printrun #806: .import error

soosalameh, 2017-02-23:
Hello,

I'm editing the pronterface.py file in the printrun folder. When I tried to build and run the file, I am getting an error related to the imports in the beginning of the file, like in line 32 "from . import pronsole"

The error I'm getting says:
"  from . import pronsole
ValueError: Attempted relative import in non-package"

Does anyone know how to fix this?

rockstorm101, 2017-05-07:
> The error I'm getting says:
" from . import pronsole
ValueError: Attempted relative import in non-package"

This means 'pronterface.py' file looks for a file named 'pronsole.py' which should be in the same folder. It seems you moved either of them and now they are not in the same folder.

rockstorm101, 2017-12-13:
Hi, @soosalameh, did my answer solve your problem? If so, could you please close this issue?

rockstorm101, 2018-02-24:
I'm closing this issue. The question was answered and the OP did not complain. Please feel free to reopen if needed.

closed 2018-02-24 by rockstorm101